indilib/indi #2051: Error when compiling sources from repo

kris969, 2024-05-01:
I tried to compile sources from git repo, strictly following the howto available.

Two weeks ago it was possible to compile all the drivers on the same context. today it was not!  

**Expected behavior**
a full make and install process

**Desktop (please complete the following information):**
I am using an freshly installed Debian bookworm on a Raspi 4.
 - OS: Debian bookworm
 - Version : latest available

**Log Files**
Here is the output showing the error messages : 

[ 17%] Built target indiclientstatic
[ 18%] Linking CXX shared library libindiclient.so
/usr/bin/ld: /usr/lib/aarch64-linux-gnu/libnova.a(solar.o): relocation R_AARCH64_ADR_PREL_PG_HI21 against symbol `ln_get_solar_equ_coords' which may bind externally can not be used when making a shared object; recompile with -fPIC
/usr/lib/aarch64-linux-gnu/libnova.a(solar.o): in function `ln_get_solar_rst_horizon':
(.text+0x268): dangerous relocation: unsupported relocation
/usr/bin/ld: /usr/lib/aarch64-linux-gnu/libnova.a(solar.o): relocation R_AARCH64_ADR_PREL_PG_HI21 against symbol `ln_get_solar_equ_coords' which may bind externally can not be used when making a shared object; recompile with -fPIC
/usr/lib/aarch64-linux-gnu/libnova.a(solar.o): in function `ln_get_solar_rst':
(.text+0x280): dangerous relocation: unsupported relocation
collect2: error: ld returned 1 exit status
make[2]: *** [libs/indiclient/CMakeFiles/indiclient.dir/build.make:141: libs/indiclient/libindiclient.so.2.0.7] Error 1
make[1]: *** [CMakeFiles/Makefile2:1944: libs/indiclient/CMakeFiles/indiclient.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....

knro, 2024-05-02:
Looks like upstream libnova needs to be released with -fpic ?

kris969, 2024-05-02:
Thanks knro for this remark. 
On my side, if I understand, I just have to wait this next upcoming upstream to be taken in account by debian team, then after a debian upgrade it will be ok. That's it?

kris969, 2024-05-13:
Update...
The last version Ubuntu server 24.04 LTS 64 bits is now available from rpi-imager. I have just done a test on a fresh install, with this setup I was able to compile and install the indi core repo without any error.
It's up to you to close this issue.

closed 2024-05-13 by knro