indilib/indi #2070: Add Pulse Guiding to LX200 Classic Driver over Serial Port

windows2000bug, 2024-06-02:
Currently, the Indi driver does not support Pulse Guiding over the RS232 port like ASCOM does for the LX200 Classic telescope mount. I spoke to an ASCOM developer and looked at the code and all they are doing for the classic is telling the scope todo to the following:

Direction = North, East, South, or West (N,E,S,W in command)
1) Slew to <direction> ( :M<direction># ) 
2) delay(pulseDuration*10) //delay for the pulse duration in MS
3) Stop Slewing to <direction>( :Q<direction># )

I have a Hardware work around, though not everyone knows how to solder. I am parsing out the ":Mg" command that is being sent to the scope that the LX200 ignores via an Arduino and some other hardware.

knro, 2024-07-03:
This is what **manual** pulse guiding is. Not sure how reliable this is. Did you implement it in your branch and tested it out with guiding?

windows2000bug, 2024-08-20:
> This is what **manual** pulse guiding is. Not sure how reliable this is. Did you implement it in your branch and tested it out with guiding?

I am not a developer by any means, just a hobbiest who knows enough to be dangerous :)

I don't have any branches for Indi nor the LX200 driver. I just created a small prototype HW to hijack the :Mg command that is being sent to the LX200 Classic.

I am hoping someone that understands the IndiCode as well as more experience as a developer to implement this.

closed 2024-10-28 by github-actions[bot]