indilib/indi #2081: After KStars last update, indi_scopedome_dome driver crash

Carlos-Vera-H, 2024-07-11:
**Describe the bug**
After last update of KStars to version 3.7.1, **indi_scopedome_dome driver crash** (the other indi device drivers in my Ekos profile works properly)

The ScopeDome is Arduino version. After the crash, I have tested it in Windows, with Ascom driver, and works properly. 
______________________________________________________________________________________________________
**To Reproduce**
1. Start Kstars
2. Start Ekos (INDI client)
3. Execute any Ekos profile including indi_scopedome_dome driver.

4. OBSERVED RESULT:
All INDI services start properly, but indi_scopedome_dome driver crashes inmediately. Then it try to restart and crashes again and again ... (when on a very rare occasion, it has started normally, its operation remains normal and without interruptions.)

As you can see in the attached log files, the connection is established correctly, but the driver always fails in some step of its initialization.
______________________________________________________________________________________________________
**Expected behavior**
That indi_scopedome_dome driver works properly, like before last update of KStars.
______________________________________________________________________________________________________
**Desktop (please complete the following information):**
 - OS: Linux Mint:
 - Version: 21.2 
 - Desktop: Cinnamon 5.8.4
 
- INDI driver: indi_scopedome_dome 
- version 2.0
- interface 32
 
- KStars Version 3.7.1 Stable  2024-06-18T09:22:30Z
______________________________________________________________________________________________________
**Screenshots**
If applicable, add screenshots to help explain your problem.
______________________________________________________________________________________________________
**Log Files**

- INDI driver log (indi_scopedome_dome)
[indi_scopedome_dome_10:59:40.log](https://github.com/user-attachments/files/16182470/indi_scopedome_dome_10.59.40.log)

- KStars-Ekos log, including several attempts to execute indi_scopedome_dome driver 
[log_21-28-59.txt](https://github.com/user-attachments/files/16182441/log_21-28-59.txt)

TallFurryMan, 2024-07-14:
Thanks for this report. The interesting part of the (snipped) log is:
```
2024-07-09T21:29:45 - INDI Server:  "2024-07-09T20:29:45: Driver indi_scopedome_dome: Impossible ISState 96"
2024-07-09T21:29:45 - INDI Server:  ""
2024-07-09T21:29:45 - ScopeDome Dome :  "[INFO] Steps per revolution read as 3280 "
2024-07-09T21:29:45 - ScopeDome Dome : "[DEBUG] Dome position 7.804878, step count 32000 "
2024-07-09T21:29:45 - INDI Server:  "2024-07-09T20:29:45: Driver indi_scopedome_dome: read: Connection reset by peer"
2024-07-09T21:29:45 - INDI Server:  "<delProperty device=\"ScopeDome Dome\"/>"
2024-07-09T21:29:45 - INDI Server:  "2024-07-09T20:29:45: Driver indi_scopedome_dome: Terminated after #0 restarts."
2024-07-09T21:29:45 - INDI Server:  ""
2024-07-09T21:29:45 - Removing device "ScopeDome Dome"
2024-07-09T21:29:45 - INDI driver  "indi_scopedome_dome"  crashed!
```

What is the version of your indi package installation? Would you care to test with indi 2.0.8?

Carlos-Vera-H, 2024-07-14:
The INDI library used in the referred test, where the SopeDome driver failure occurs, is 2.0.8:

INDI library: 2.0.8
Code 2.0.8-tgz. Protocol 1.7. 

[image]

knro, 2024-07-15:
Can you run the driver in [Ekos Debugger](https://www.stellarmate.com/support/tutorials/ekos-debugger.html) and submit the backtrace?

Carlos-Vera-H, 2024-07-15:
Here you have two different reports, generated by Ekos Debugger:

[indi_logs_24-07-15T17-53-34.zip](https://github.com/user-attachments/files/16238524/indi_logs_24-07-15T17-53-34.zip)
[indi_logs_24-07-15T18-03-14.zip](https://github.com/user-attachments/files/16238528/indi_logs_24-07-15T18-03-14.zip)

Thanks

Carlos-Vera-H, 2024-07-17:
[Snip_of_KStars-Ekos_log_21-28-59.txt](https://github.com/user-attachments/files/16273342/Snip_of_KStars-Ekos_log_21-28-59.txt)

As we can see in the attached file (a commented clipping of the file generated by KStars), in the indi_scopedome_dome driver, there are two failures:
 
- The first and least important one, exists from the first release of 2.0 version of the driver that I’ve used before, and occurs almost always on the first connection attempt as a "handshake" failure, which usually resolves on the second try, as on this occasion.

- The second bug is the most important one, that appeared in the latest Kstars update, and occurs after the connection is successfully established, always in the same step of the initialization sequence (as I have already mentioned previously).

knro, 2024-07-18:
Try deleting the config files under ~/.indi and restart the driver. Any better results? Also, do you have the Dome Intertia file that was referenced?

Carlos-Vera-H, 2024-07-18:
I just did, several times, what you suggested (delete the config files), but the result has not changed at all (the driver is interrupted at the same point in the sequence).

... And yes, I have the inertia table generated by the Windows application provided by the 
manufacturer (copied in the linux .../.indi folder) and it has always worked well.
[ScopeDome_DomeInertia_Table.txt](https://github.com/user-attachments/files/16282730/ScopeDome_DomeInertia_Table.txt)

Carlos-Vera-H, 2024-07-25:
A provisional work around to this issue will be to revert to the previous version of the driver, which at least worked fine on the second try.
  
Can someone tell me, please, where and how can I get and install it?.

Thank you.

Carlos-Vera-H, 2024-08-05:
Some tests I have done on the INDI ScopeDome driver:

    1.- Whitout uninstalling anything, I have compiled  the package libindi_2.0.2+202305311719~ubuntu22.10.1.tar.xz
       and installed it by “cmake install”.
       Result: the  ScopeDome driver always works fine on the second try, but fails the Zwo Asi cameras and the Starlight Xpress AO drivers.
       
    2.- I have compiled  the package libindi_2.0.5+202312021714~ubuntu23.04.1.tar.xz
       and installed it by “cmake install”.
       Result: the  ScopeDome driver always works fine on the second try, but fails the Zwo Asi cameras and the Starlight Xpress AO drivers.
       
    3.- I have completely uninstalled indi-full and kstars-bleeding, then I have updated it to the last version (with indi-full 2.0.9).
       Result: only the ScopeDome driver almost always fails (the errors mentioned are still there). All other drivers works fine again.

Can you tell me how can I do a complete installation of an early version of this software? 
(when I try it in the conventional recommended way, the system always installs the last version).

Thank you.

Carlos-Vera-H, 2024-08-06:
By QT Creator > Tools > Diff > Diff external files, I have compared the source files of ScopeDome driver versions 2.0.5 and 2.0.8, and I have found a lot of differences only in the file scopedome_dome.cpp; so I think that might be where the bug is.

[image]

There is no difference between versions 2.0.8 and 2.0.9

knro, 2024-08-07:
I think it is best if install Rustdesk and then I can remotely login to debug this. Send the details to my email (mutlaqja AT ikarustech DOT com)

Carlos-Vera-H, 2024-08-09:
I've installed Rustdesk and send the details to your email.
Thank you.

Carlos-Vera-H, 2024-08-14:
After a rebuild and reinstall of indi core and drivers, made by knro, I've tested the driver, connecting and disconnecting it a few times, and it has always started well. So I think that now the bug is fixed. 

(Only remains the mentioned failure of connection in the first attempt, which is solved on the second attempt without any crash of the driver; but this is another less important issue).

Thank you very much. 

Best regards 

Carlos Vera Hernández.

closed 2024-08-14 by Carlos-Vera-H