indilib/indi #2083: Add indi-ipx800v4 Driver

aknotwot, 2024-07-21:
No other driver were found. 
Driver is stored in [GitHub](https://github.com/aknotwot/indi-ipx800v4)
It's a drive made first to manage the roof. It's a part of Dome device family

The driver communication with IPX800v4 comands through TCP.  
[IPX800 API](https://wiki.gce-electronics.com/index.php?title=API_V4#Liste_des_commandes)

knro, 2024-07-25:
Thank you. I think this driver needs to be broken down to two drivers:
* A Relay-only driver
* A Roll Off Roof driver

The relay driver would be based on INDI::RelayInterface that I just introduced a few days. I can do this and migrate this part of the driver to use IPX800. Then I plan to implement a universal ROR driver that uses the relay driver.

How do you get the limit switch status? via digital input from the same IPX800? I think I need to create another INDI::InputInterface as well.

aknotwot, 2024-07-25:
Hi,
about the limit : in my situation the roof's engine detect by it self the limit. Anyway i added digital inputs to check if roof is opened or closed.
Relay only would be dedicated to the IPX800, the Roll Off Roof driver dedicated to the function affected to the IPX800 ? I agree, currently this driver is too much  dependant of my own configuration.
I started to rewrite the driver using your new RollOff Ino driver. But it's realy early version.

Can I help the migration ? 
Best Regards,

knro, 2024-07-26:
My current plan is this:

* INDI::OutputInterface: This is for digital outputs (On/Off) like Relays and GPIOs.
* INDI::InputInterface: This is for digital/analog inputs like GPIOs and some observatory controllers (like the IPX800)
* INDI::ROR driver that you set in it which drivers you want to use as input and output

In the ROR driver you specify the numbers for each. For example, assuming IPX800, we would put this driver as both input and output driver. Then in ROR  we say this:
* Roof Open: DIGITAL_OUTPUT_2,DIGITAL_OUTPUT_3 --> this will send command to output driver to trigger outputs 2 and 3
*  Roof Close: DIGITAL_OUTPUT_1
* Roof Opened: DIGITAL_INPUT_5
* Roof Closed: DIGITAL_INPUT_6

Would this work for your case? Btw, the ROR driver is *not written yet*, it will be next on my TODO after I finish the new indi-gpio driver.

knro, 2024-07-29:
@aknotwot The INDI::InputInterface and INDI::OutputInterface are now ready to be used. I added [Waveshare modbus](https://github.com/indilib/indi/blob/master/drivers/auxiliary/waveshare_modbus_relay.h) driver that uses the Output interface, and INDI [GPIO](https://github.com/indilib/indi-3rdparty/tree/master/indi-gpio)  driver that uses both Input and Output interfaces. Do you think you can modify your driver to use these interfaces?

aknotwot, 2024-08-01:
> @aknotwot The INDI::InputInterface and INDI::OutputInterface are now ready to be used. I added [Waveshare [...]
Hi.
I can. But not before second part of august. Holidays are starting !

knro, 2024-08-18:
@aknotwot Let me know if you require any additional help to integrate this driver.

knro, 2024-09-12:
@aknotwot Any updates on this driver? We're getting close to INDI 2.1.0 release.

aknotwot, 2024-09-12:
> @aknotwot Any updates on this driver? We're getting close to INDI 2.1.0 release.

Hi, didn't work on it yet. Will try this weekend.  I give you updates sunday evening.

aknotwot, 2024-09-14:
Hi Jasem, starting to work on it. Not sure too understand the full new architecture .

knro, 2024-09-14:
Did you take a look at the INDI [GPIO](https://github.com/indilib/indi-3rdparty/tree/master/indi-gpio) driver? This is a good example on how to implement both Input and Output interfaces. Please let me know if you have any questions. We could do a quick meeting as well if anything is not clear yet.

aknotwot, 2024-09-14:
Should be ok with the two examples (GPIO and waveshare). Thx.

aknotwot, 2024-09-18:
> Should be ok with the two examples (GPIO and waveshare). Thx. 

Hi Jasem, i won't be able to finish update of my driver in the next 2 weeks. I you were, perhaps you shouldn't wait for them to publish the new indi release. I will be ready for next. 

Best Regards,

aknotwot, 2024-10-06:
Hi Jasem. Back on track. Work in progress. Trying first to implement new Input/Output Interfaces, then will see how to use ROR. Regards,

knro, 2024-10-07:
@aknotwot Great! A couple of users are waiting for this driver! Let me know if you need any help.

aknotwot, 2024-10-08:
Hi @knro . Some news. Hard to fully understand the architecture. Mostly don't understand how to use universal_ror driver...

knro, 2024-10-09:
@aknotwot Can you please read the [ROR Guide](https://indilib.org/develop/tutorials/233-universal-roll-off-driver.html)? Let me know if anything is not clear.

knro, 2024-10-09:
What do you mean an IP to universal ROR? Please clarify. You can also join us on [Discord](https://discord.gg/xhgxcqaC) if you want faster response other than here.

aknotwot, 2024-10-10:
@knro, i'm struggling to add an IP adress to the IPX driver , tried many things ...connection is failing... 
Can you tell what is wrong in this example, or perhaps a better way to add an IP manager in IPX800 as a defaultDevice.

Thanks for you help.

`     IPPortTP.update(texts, names, n);
		IPPortTP.setState(IPS_OK);
		IPPortTP.apply();
		saveConfig(IPPortTP);
		LOG_DEBUG("Updating Connection - End1 ");
		ipAddress = IPPortTP[0].text;
		myPortFD= atoi(IPPortTP[1].text);
		LOG_DEBUG(ipAddress);
		LOG_DEBUG(IPPortTP[1].text);
		LOG_DEBUG("Updating Connection - Starting");
		tcpConnection = new Connection::TCP(this);
		tcpConnection->setConnectionType(Connection::TCP::TYPE_TCP);
		/*tcpConnection->setDefaultHost(ipAddress);
		tcpConnection->setDefaultPort(myPortFD);*/
		tcpConnection->setDefaultHost("192.168.1.220");
		tcpConnection->setDefaultPort(9870);
		
		LOG_DEBUG("Updating Connection - Handshake");
		tcpConnection->registerHandshake([&]()
		{
			LOG_DEBUG("Updating Connection - Call Handshake");
			return Handshake();
		}
		
		);
		registerConnection(tcpConnection);`

knro, 2024-10-11:
Do not define your own property, use the tcpCOnnection. Let me see if I can make the driver migration easy for you.

aknotwot, 2024-10-11:
Hi, i will join Discord.
Ip is linked to the ipx800 (or any other controler), why we have to choose IP in ROR driver ?
Recreate text fields in IPX800 was the only way i found to display fields to choose an IP and port.

As you can see i'm not fluent at all in INDI architecture...

I have a draft driver, but can't test without operationnal connection.
Best regards.

-------- Message d'origine --------

knro, 2024-10-11:
Ok I took a look at the driver and it needs substantial changes. I already started on a skeleton driver for IPX800 but need to know the following:
1. It appears from documentation an API key is used, not username and password?
2. We need Digital Inputs, Analog Inputs, and Digital Outputs.
3. Which commands used for DI, AI, and DO exactly? GetR to get all digital output status? I looked up the documentation but it was in French.

aknotwot, 2024-10-11:
Hi, wait a minute, last minutes were efficient. I removed my mess in the code about ip field adress, just left the new TcP...so now ip connection is working. I'll give more news in the weekend ! I hope to send you a working draft in the weekend.

-------- Message d'origine --------

aknotwot, 2024-10-13:
Hi Jasem,
the IPX part looks fonctionnal, at least enough to test integration with Input/Output Interface and universal_ROR...
In IPX driver Input/Output Interface are updated correctly.

But clearly, integration with universe_ROR is not working, as with the Dome module. Reading codes i don't see what is missing...I don't understand how universal_ROR can activate IPX800 Relay through OutputInterface.

Here you can find the last version : https://github.com/aknotwot/indi-ipx800v4/tree/dev
Thanks for your help.
Best Regards,
Arnaud D.

Envoyé avec la messagerie sécurisée [Proton Mail.](https://proton.me/mail/home)

aknotwot, 2024-10-28:
Hi @knro still working on the topic. I modify the name of the driver : indi-ipx800, allowing it to be more generic cf here : https://github.com/aknotwot/indi-ipx800 . Later i will implement V3, V5 commands.  Integration with Input/Outpout Interface work well. But still more difficulties to make universal ror behave correctly. I can't test my roof in real condition because of bad weather ... :-/

knro, 2024-11-23:
@aknotwot Can you please submit the driver to INDI core (not 3rd party)?

aknotwot, 2024-11-23:
> @aknotwot Can you please submit the driver to INDI core (not 3rd party)?

Hi jasem, yes sorry, busy days. Will look at it tomorrow 👍

closed 2025-01-25 by knro