indilib/indi #2089: ZWO AM5 Mount - Reset home position [I have a solution]

pawel-soja, 2024-07-27:
**Is your feature request related to a problem? Please describe.**
I have a ZWO AM5 mount, I know that by sending the appropriate command, you can save a new home position in the mount. The problem often appears on forums, where a newly purchased AM5 has a position shifted by a few degrees.
For example: https://www.cloudynights.com/topic/886794-zwo-am5-help-with-home-position/

**Describe the solution you'd like**
The problem can be solved by using other software, but I wish such functionality was available with KStars/EKOS.
I can add a button to the lx200am5 driver, but I'm not sure where.

**Describe alternatives you've considered**
I can't imagine anything outside of KStars :heart: 

**Additional context**
I found these buttons, I've never used them and I'm not sure if this is the right place. I just want to send the appropriate command by pressing the button. Mount remembers and considers this as the starting position.
[image]

Where can I implement resetting the home position?

knro, 2024-07-28:
There is [Homing support](https://github.com/indilib/indi/blob/ad1175385baca4d0001a5dc6f114d9c9544eb923/drivers/telescope/lx200am5.cpp#L97) in the AM5 driver already, so there needs to be another one to set as-is and it would accept whatever current position as home?

pawel-soja, 2024-07-28:
I see that big changes are coming: https://github.com/indilib/indi/pull/2090

For example, I was thinking about something like this:
[image]
1. First, to do something, you need UnPark
2. Driving to home position, without parking, so mount returns correct position (opto sync)
3. You set the correct position. The new position cannot be too far away, or the process will fail.
4. Saving position in mount.

I would extend `HomeSP`, but I see that your latest modifications are causing a conflict.
```cpp
HomeSP[GoHome].fill("GO", "Go", ISS_OFF);
HomeSP[SaveHome].fill("SAVE", "Save", ISS_OFF);
HomeSP.fill(getDeviceName(), "TELESCOPE_HOME", "Home", MAIN_CONTROL_TAB, IP_RW, ISR_ATMOST1, 60, IPS_IDLE);

/* ... */

switch (HomeSP.findOnSwitchIndex())
{
    case GoHome:
        LOG_INFO("Homing in progress...");
        HomeSP[GoHome].setState(ISS_ON);
        HomeSP.setState(goHome() ? IPS_BUSY : IPS_ALERT);
        break;
    case SaveHome:
        if (saveHome())
        {
            HomeSP.setState(IPS_OK);
            LOG_INFO("Current position written as home position in the mount.");
        }
        else
        {
            HomeSP.setState(IPS_ALERT);
            LOG_WARN("Failed to write current home position in the mount. It is possible that the position is too far from the factory position.");
        }
        HomeSP[SaveHome].setState(ISS_OFF);
        break;
}
```

knro, 2024-07-28:
For this, HOME_SET and HOME_GO can be used. I already modified the driver in #2090 so you just need to add saveHome() there.

pawel-soja, 2024-07-29:
Mission complete!
[image]

Thanks Jasem!

closed 2024-07-29 by knro