indilib/indi #2096: indi-asi cannot expose shorter than 2 seconds (with an asi1600 camera)

d33psky, 2024-08-10:
**Describe the bug**
indi-asi (2.4+t202408021338~ubuntu22.04.1) can no longer make exposures shorter than 2 seconds with an asi1600 camera.

**To Reproduce**
Exact steps to reproduce the behavior.
1. set up indi-asi driver with asi1600 camera
2. try to make a photo shorter than 2 seconds, either in the ekos camera tab or in the focuser routine
3. observe [ERROR] Exposure failed after 3 attempts

**Expected behavior**
The same behaviour as a longer-than-2-seconds exposures. For example : 
[INFO] Taking a 60 seconds frame...
[INFO] Exposure done, downloading image...
[INFO] Download complete.

**Forum discussion link**
https://indilib.org/forum/ccds-dslrs/14336-zwo-camera-failing-exposure.html

**Desktop (please complete the following information):**
 - Ubuntu 22.04.4 LTS
 - indi-asi (2.4+t202408021338~ubuntu22.04.1)

**Log Files**
last part from [log_14-02-43.txt](https://github.com/user-attachments/files/16569472/log_14-02-43.txt)

[2024-08-10T14:03:27.538 CEST INFO ][   org.kde.kstars.ekos.capture] - "Job requires 1,000-second Luminance images, has 0/1 frames captured and will be processed."
[2024-08-10T14:03:27.548 CEST DEBG ][   org.kde.kstars.ekos.capture] - Focus elapsed time (secs):  0 . Requested Interval (secs):  1800
[2024-08-10T14:03:27.553 CEST INFO ][   org.kde.kstars.ekos.capture] - "Capture exposure = 1 sec, type = Light, filter = Luminance, upload mode = 0, batch mode = true, seq prefix = _%t_%T_%F_%D_%t_%T_%F_%e_Light_Luminance_1_secs_ISO8601, gain = 100, offset = 60, format = Raw 16 bit, encoding = FITS, binning = 1x1, ROI = (0+4656, 0+4656)"
[2024-08-10T14:03:27.553 CEST DEBG ][   org.kde.kstars.ekos.capture] - Capture State changes from "In Progress" to "Capturing"
[2024-08-10T14:03:27.555 CEST INFO ][   org.kde.kstars.ekos.capture] - "Capturing 1,000-second Luminance image..."
[2024-08-10T14:03:39.694 CEST INFO ][           org.kde.kstars.indi] - ZWO CCD ASI1600MM-Cool :  "[ERROR] Exposure failed after 3 attempts. "
[2024-08-10T14:03:39.694 CEST INFO ][   org.kde.kstars.ekos.capture] - "Capture failed. Check INDI Control Panel for details."
[2024-08-10T14:03:39.696 CEST INFO ][   org.kde.kstars.ekos.capture] - "Restarting capture attempt #1"
[2024-08-10T14:03:39.701 CEST INFO ][   org.kde.kstars.ekos.capture] - "Capture exposure = 1 sec, type = Light, filter = Luminance, upload mode = 0, batch mode = true, seq prefix = _%t_%T_%F_%D_%t_%T_%F_%e_Light_Luminance_1_secs_ISO8601, gain = 100, offset = 60, format = Raw 16 bit, encoding = FITS, binning = 1x1, ROI = (0+4656, 0+4656)"
[2024-08-10T14:03:39.701 CEST INFO ][   org.kde.kstars.ekos.capture] - "Capturing 1,000-second Luminance image..."
[2024-08-10T14:03:39.704 CEST INFO ][   org.kde.kstars.ekos.capture] - "Capture failed. Check INDI Control Panel for details."
[2024-08-10T14:03:39.706 CEST INFO ][   org.kde.kstars.ekos.capture] - "Restarting capture attempt #2"
[2024-08-10T14:03:39.709 CEST INFO ][   org.kde.kstars.ekos.capture] - "Capture exposure = 1 sec, type = Light, filter = Luminance, upload mode = 0, batch mode = true, seq prefix = _%t_%T_%F_%D_%t_%T_%F_%e_Light_Luminance_1_secs_ISO8601, gain = 100, offset = 60, format = Raw 16 bit, encoding = FITS, binning = 1x1, ROI = (0+4656, 0+4656)"
[2024-08-10T14:03:39.710 CEST INFO ][   org.kde.kstars.ekos.capture] - "Capturing 1,000-second Luminance image..."
[2024-08-10T14:03:41.650 CEST INFO ][           org.kde.kstars.indi] - Dome Scripting Gateway :  "[INFO] Dome already parked. "
[2024-08-10T14:03:50.541 CEST INFO ][           org.kde.kstars.indi] - ZWO CCD ASI1600MM-Cool :  "[ERROR] Exposure failed after 3 attempts. "
[2024-08-10T14:03:50.541 CEST INFO ][   org.kde.kstars.ekos.capture] - "Capture failed. Check INDI Control Panel for details."
[2024-08-10T14:03:50.542 CEST INFO ][   org.kde.kstars.ekos.capture] - "Capture failed. Check INDI Control Panel for details."
[2024-08-10T14:03:50.544 CEST DEBG ][   org.kde.kstars.ekos.capture] - Capture State changes from "Capturing" to "Aborted"

knro, 2024-08-10:
Is this with SDK v1.34 or 1.35?

d33psky, 2024-08-10:
[image]
looks to be 1.35

This is the indi-asi driver from
```
deb https://ppa.launchpadcontent.net/mutlaqja/ppa/ubuntu/ jammy main
```

knro, 2024-08-10:
I triggered a rebuild so that the 1.34 can be reinstalled over 1.35

https://code.launchpad.net/~mutlaqja/+recipe/libasi-stable

d33psky, 2024-08-10:
manually downloaded and installed the resulting deb file. Issue is confirmed fixed \o/

Do we have a channel to inform ZWO ? Or is the issue in indi-3rdparty ? (so that 1.36 does not bite again)

daleghent, 2024-08-10:
FWIW, we've seen two ASI1600 users complain about this same problem on the NINA Discord. This is with SDK version 1.35 on Windows, of course.

d33psky, 2024-08-10:
Thanks for the info @daleghent , I've also heard about a Nina user (via the Libera ##astroimaging channel) with this issue. So it's in the ZWO code and not in indi-3rdparty code. -> closing this ticket.

closed 2024-08-10 by d33psky