indilib/indi #2097: Cannot do observations with kstars on multiple targets with OnStepX mount

apalazzi, 2024-08-17:
SUMMARY
When you setup a session in kstars session scheduler and the mount is OnStepX (but I think it's the same with OnStep) only the first target is captured and the following targets abort with an error.
What happens is that when Kstars sends the command to center the second (or third or fourth) target the mount refuses because is still in tracking mode and the session ends with an error.
I filed a bug in kstars https://bugs.kde.org/show_bug.cgi?id=480351 and the author replied that it's a bug in INDI, so here I am.

STEPS TO REPRODUCE
In kstars/ekos
1. setup the scheduler to capture two or more targets
2. start the scheduler
3. wait for the first target to finish and the second to start

OBSERVED RESULT
The scheduler ends with error after the first target is captured.

EXPECTED RESULT
The scheduler should continue capturing all the targets.

SOFTWARE/OS VERSIONS
KDE Frameworks Version: 5.107. 0 
Qt Version: 5.10.0

ADDITIONAL INFORMATION
OnStepX https://github.com/hjd1964/OnStepX

knro, 2024-08-18:
Can you please enable debug logging and attach logs?

apalazzi, 2024-08-18:
Just to be sure: you mean the debug logging in kstars/ekos, right?

knro, 2024-08-19:
Yes, make sure the INDI+Mount checkboxes are selected.

apalazzi, 2024-10-20:
I've since switched from kstars to CCDCiel and had not this issue; not sure I'll have time in the future to further test this.

closed 2024-12-29 by github-actions[bot]