indilib/indi #2102: OnStepX running on MaxESP4 fails to connect

gy-lehel, 2024-08-22:
**Describe the bug**
OnStepX running on some MaxESP4 boards (tested on JTW Manticore with latest OnStepX release, and also with latest OnStepX build from git repo) starts up in a state that blocks handshaking with the mount over the TTYUSB port.

Writing a small utility that is sending the ":GVP#" command at least one time on the serial link clears the buffer, and allows indi to connect to the mount. 

**To Reproduce**
Exact steps to reproduce the behavior.
1. Start up OnStepX on a maxESP board (JTW Manticore)
2. Try to connect mount
3. After timeout mount fails to connect
4. Send :GVP# command using a tty terminal, receive '0' as response, send again, receive 'OnStep' response
5. Try to connect mount
6. Works as expected

**Expected behavior**
Mount connected without manual workaround

**Screenshots**
N/A

**Desktop (please complete the following information):**
 - OS: StellarMateOS
 - Version 2.0.x (all versions)

**Log Files**
As agreed with Jason I started working on a workaround in indi, this issue is to cover the related code changes.

gy-lehel, 2024-08-22:
[0001-Add-workaround-for-OnStepX-failing-connection.txt](https://github.com/user-attachments/files/16712938/0001-Add-workaround-for-OnStepX-failing-connection.txt)

knro, 2024-08-22:
Thanks, can you please submit a PR?

closed 2024-09-12 by knro