indilib/indi #2108: Usability Request - Scheduler

stevecoh1, 2024-09-07:
**Is your feature request related to a problem? Please describe.**
I was using Ekos to schedule a job to run in the future.  I used the CCD module to schedule a job.  Everything looked good and I wanted to save it.  For several minutes I could not get Ekos to accept the job into the queue.  I was pounding my head trying to figure what the problem was.  Was it a WiFi connectivity issue? No.  Was it something wrong with my inputs?  It all looked good.  Then it occurred to me: I had typed in my target as "M32". It should have been "M 32". Making that change, allowed the job to be added to queue and saved.

**Describe the solution you'd like**
An error message!  This is like an online form that won't let you continue unless you get all the inputs right.  Such forms ALWAYS show an error message like "Invalid Credit Card Expiration Date" or something.  In this case, it could be "Target unknown" or something of that sort.  Since it is not infrequent to have connectivity issues it would be nice to know that Ekos has received my request and found it wanting, rather than a connectivity issue.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
[image]

knro, 2024-09-08:
This is not an INDI issue.

Nevertheless, if you used the Find Dialog to select the tool, then the coordinates fields would be automatically filled. Your issue is not the name, it's that there are no J2000 RA/DE specified.

closed 2024-09-08 by knro

stevecoh1, 2024-09-10:
Sorry, but I disagree.  Yes, I know it was my mistake not picking from the list.  But the interface does allow manual entry of the target, and if the user does so and enters it wrong, he deserves an error message, not a silent failure.  Silent failures are bad UI.

knro, 2024-09-11:
You can enter manually, but then you'd also have to enter the RA/DE manually as well.

stevecoh1, 2024-09-11:
OK, but you're missing my point: "Silent failures are bad UI".  If program lets the user do something and it's the wrong thing, he shouldn't need to scratch his head and say "WTF is wrong now?"  The higher you make learning curves, the more likely it is that users will give up or find a product that works better.