indilib/indi #2119: Lost of joystick control for telescope

hiro3110i, 2024-09-22:
**Describe the bug**
I'm not sure when it started, but it seems that since the combination of indi v2.0.9 and KStars v3.6.2 was used, joystick control has not worked.

**To Reproduce**
1. Run Ekos from KStars with Telescope Simulator, CCD Simulator and Joystick.
2. Start and Connect to indiserver on Ekos (with driver settings that have been proven to work in the previous version).
3. Can't use joystick to slew telescope.

I'm using JoyCon(L) of Nintendo Switch because it is small and easy to connect with Raspberry Pi4/5 via bluetooth.

**Expected behavior**
In the previous version, I could slew telescope using JoyCon(L), but now...

**Screenshots**
No screenshots

**Desktop (please complete the following information):**
 - OS: 
       Ubuntu 24.04 on Raspberry Pi5
       Ubuntu Mate 22.04 on Raspberry Pi4
 - Server and Clients: indi v2.0.9 and KStars v3.6.2
       which is installed by "sudo apt install indi-full kstars-bleeding".

**Log Files**
No log files but we can see the warning message on Ekos.
`[WARNING] Motion control is set to unknown value 1!`

**Further Investigation**
Joystick control would come back if I changed the code $(repo_dir)/libs/indibase/inditelescope.cpp like below.
Could you please review the diff ?
```
1540c1540
<             if (MotionControlModeTP[MOTION_CONTROL_JOYSTICK].getState() == ISS_ON)
---
>             if (MotionControlModeTP[MOTION_CONTROL_MODE_JOYSTICK].getState() == ISS_ON)
1542c1542
<             else if (MotionControlModeTP[MOTION_CONTROL_AXES].getState() == ISS_ON)
---
>             else if (MotionControlModeTP[MOTION_CONTROL_MODE_AXES].getState() == ISS_ON)
2425c2425
<     if (MotionControlModeTP[MOTION_CONTROL_JOYSTICK].getState() == ISS_ON && !strcmp(joystick_n, "MOTIONDIR"))
---
>     if (MotionControlModeTP[MOTION_CONTROL_MODE_JOYSTICK].getState() == ISS_ON && !strcmp(joystick_n, "MOTIONDIR"))
2441c2441
<     if (MotionControlModeTP[MOTION_CONTROL_AXES].getState() == ISS_ON)
---
>     if (MotionControlModeTP[MOTION_CONTROL_MODE_AXES].getState() == ISS_ON)
```
I couldn't understand the difference between the following enums in inditelescope.h.
```
        // Switch for choosing between motion control by 4-way joystick or two separate axes
        INDI::PropertySwitch MotionControlModeTP {2};
        enum
        {
            MOTION_CONTROL_MODE_JOYSTICK,
            MOTION_CONTROL_MODE_AXES,
            MOTION_CONTROL_JOYSTICK,
            MOTION_CONTROL_AXES
        };
```

I have one more question about the slew motion of telescope.
The East/West slew rate changes depending on the North/South position.
Especially near the North Pole, the East/West slew rate is very small.
Is this the intended specification? Could you please confirm it?

knro, 2024-09-22:
@naheedsa Please check if we have a regression

hiro3110i, 2024-09-24:
@naheedsa Thank you very much for your quick update.
Joystick slewing is restored by [https://github.com/indilib/indi/pull/2120](url) and [https://github.com/indilib/indi/pull/2123](url).

closed 2024-09-24 by hiro3110i