indilib/indi #2122: indi-apogee crashes when connecting Apogee Alta F47 with V22 firmware

Rakaanishu, 2024-09-24:
The indi-apogee crashes when connecting Apogee Alta F47 with V22 firmware. This was apparently known as it is mentioned by a review on the driver download page but as the camera is not supported anymore by Andor instruments, there is no mean to upgrade its firmware.

**To Reproduce**
1. Just try to connect the camera on Ekos

**Expected behavior**
The camera should be connected but the log says that the driver has crashed.

**Desktop (please complete the following information):**
 - OS: Ubuntu 22.04.5 LTS 64 bits
 - Version: Indi Lib 2.0.9

Rakaanishu, 2024-09-26:
From Ekos Debugger

Debug log
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[Attaching after Thread 0x7ffff7f89740 (LWP 8725) fork to child process 8728]
[New inferior 2 (process 8728)]
[Detaching after fork from parent process 8725]
[Inferior 1 (process 8725) detached]
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
process 8728 is executing new program: /usr/bin/indi_apogee_ccd
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".

Application log
2024-09-26T17:19:35: startup: /usr/bin/indiserver -r 0 -v indi_apogee_ccd
2024-09-26T17:19:35: Driver indi_apogee_ccd: pid=8728 rfd=6 wfd=6 efd=7
2024-09-26T17:19:35: listening to port 7624 on fd 5
2024-09-26T17:19:35: Local server: listening on local domain at: @/tmp/indiserver
2024-09-26T17:19:36: Driver indi_apogee_ccd: snooping on Apogee CCD.FILTER_SLOT
2024-09-26T17:19:36: Driver indi_apogee_ccd: snooping on Apogee CCD.FILTER_NAME
2024-09-26T17:19:36: Driver indi_apogee_ccd: snooping on SQM.SKY_QUALITY

knro, 2024-09-27:
No crash above.. you need to use the driver until it crashes.

Rakaanishu, 2024-09-27:
Not sure I fully understand how to use Ekos Debugger. I start the indi-apogee driver from it then if I try to connect the camera from Ekos panel, I get a warning saying that an Indi server instance is already active. 

I get this from Ekos Debugger  debug log:
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[Attaching after Thread 0x7ffff7f89740 (LWP 22381) fork to child process 22384]
[New inferior 2 (process 22384)]
[Detaching after fork from parent process 22381]
[Inferior 1 (process 22381) detached]
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
process 22384 is executing new program: /usr/bin/indi_apogee_ccd
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
Thread 2.1 "indi_apogee_ccd" received signal SIGPIPE, Broken pipe.
[Switching to Thread 0x7ffff589eb80 (LWP 22384)]
0x00007ffff7714887 in __GI___libc_write (fd=2, buf=0x7fffffffabc0, nbytes=21) at ../sysdeps/unix/sysv/linux/write.c:26
#0  0x00007ffff7714887 in __GI___libc_write (fd=2, buf=0x7fffffffabc0, nbytes=21) at ../sysdeps/unix/sysv/linux/write.c:26
#1  0x00007ffff768aeed in _IO_new_file_write (f=0x7ffff781b6a0 <_IO_2_1_stderr_>, data=0x7fffffffabc0, n=21) at ./libio/fileops.c:1180
#2  0x00007ffff768b7d0 in new_do_write (to_do=<optimized out>, data=0x7fffffffabc0 "indi_apogee_ccd: EOF\n\177", fp=0x7ffff781b6a0 <_IO_2_1_stderr_>) at ./libio/libioP.h:947
#3  _IO_new_file_xsputn (n=21, data=<optimized out>, f=<optimized out>) at ./libio/fileops.c:1254
#4  _IO_new_file_xsputn (f=0x7ffff781b6a0 <_IO_2_1_stderr_>, data=<optimized out>, n=21) at ./libio/fileops.c:1196
#5  0x00007ffff76776d2 in buffered_vfprintf (s=0x7ffff781b6a0 <_IO_2_1_stderr_>, format=format@entry=0x7ffff7ed10ae "%s: EOF\n", args=args@entry=0x7fffffffcc00, mode_flags=mode_flags@entry=2) at ../libio/libioP.h:947
#6  0x00007ffff767665e in __vfprintf_internal (s=<optimized out>, format=0x7ffff7ed10ae "%s: EOF\n", ap=ap@entry=0x7fffffffcc00, mode_flags=2) at ./stdio-common/vfprintf-internal.c:1236
#7  0x00007ffff7734d13 in ___fprintf_chk (fp=<optimized out>, flag=<optimized out>, format=<optimized out>) at ./debug/fprintf_chk.c:33
#8  0x00007ffff7e1cfc7 in ?? () from /lib/x86_64-linux-gnu/libindidriver.so.2
#9  0x00007ffff7eb69cb in ?? () from /lib/x86_64-linux-gnu/libindidriver.so.2
#10 0x00007ffff7eb6e27 in eventLoop () from /lib/x86_64-linux-gnu/libindidriver.so.2
#11 0x00007ffff7e1d6f7 in main () from /lib/x86_64-linux-gnu/libindidriver.so.2
#12 0x00007ffff7629d90 in __libc_start_call_main (main=main@entry=0x7ffff7e1d5a5 <main>, argc=argc@entry=1, argv=argv@entry=0x7fffffffdf38) at ../sysdeps/nptl/libc_start_call_main.h:58
#13 0x00007ffff7629e40 in __libc_start_main_impl (main=0x7ffff7e1d5a5 <main>, argc=1, argv=0x7fffffffdf38, init=<optimized out>, fini=<optimized out>, rtld_fini=<optimized out>, stack_end=0x7fffffffdf28) at ../csu/libc-start.c:392
#14 0x000055555555a285 in ?? ()
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[Attaching after Thread 0x7ffff7f89740 (LWP 22405) fork to child process 22408]
[New inferior 2 (process 22408)]
[Detaching after fork from parent process 22405]
[Inferior 1 (process 22405) detached]
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
process 22408 is executing new program: /usr/bin/indi_apogee_ccd
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".

On Ekos side, it doesn't say that the driver crashed but didn't manage to connect the camera. Is it the correct way to proceed?

Rakaanishu, 2024-10-03:
This is from Ekos logfile, without Ekos Debugger launched, showing the crash of indi-apogee driver. 

[2024-10-03T16:51:22.542 CEST INFO ][                org.kde.kstars] - Welcome to KStars 3.7.2 Stable
[2024-10-03T16:51:22.542 CEST INFO ][                org.kde.kstars] - Build: 2024-09-06T14:55:14Z
[2024-10-03T16:51:22.542 CEST INFO ][                org.kde.kstars] - OS: "ubuntu"
[2024-10-03T16:51:22.542 CEST INFO ][                org.kde.kstars] - API: "x86_64-little_endian-lp64"
[2024-10-03T16:51:22.542 CEST INFO ][                org.kde.kstars] - Arch: "x86_64"
[2024-10-03T16:51:22.542 CEST INFO ][                org.kde.kstars] - Kernel Type: "linux"
[2024-10-03T16:51:22.542 CEST INFO ][                org.kde.kstars] - Kernel Version: "6.8.0-45-generic"
[2024-10-03T16:51:22.542 CEST INFO ][                org.kde.kstars] - Qt Version: 5.15.3
[2024-10-03T16:51:22.686 CEST DEBG ][                org.kde.kstars] - Opened the User DB. Ready.
[2024-10-03T16:51:22.782 CEST INFO ][                org.kde.kstars] - Processing  "unnamedstars.dat" , HTMesh Level 3
[2024-10-03T16:51:22.782 CEST INFO ][                org.kde.kstars] -   Sky Mesh Size:  512
[2024-10-03T16:51:22.798 CEST INFO ][                org.kde.kstars] - Loaded DSO catalog file:  "unnamedstars.dat"
[2024-10-03T16:51:22.898 CEST WARN ][                org.kde.kstars] - "Étoile HD20 794 introuvable."
[2024-10-03T16:51:22.902 CEST WARN ][                org.kde.kstars] - "Étoile HD98 230 introuvable."
[2024-10-03T16:51:22.908 CEST INFO ][                org.kde.kstars] - Loaded DSO catalogs.
[2024-10-03T16:51:22.976 CEST INFO ][                org.kde.kstars] - Loading comets
[2024-10-03T16:51:23.027 CEST DEBG ][                org.kde.kstars] - Paths to color scheme :  ("/usr/share/kstars/themes")
[2024-10-03T16:51:23.064 CEST DEBG ][                org.kde.kstars] - Reporting new timestep value:  60
[2024-10-03T16:51:23.079 CEST WARN ][                       default] - libpng warning: iCCP: known incorrect sRGB profile
[2024-10-03T16:51:23.099 CEST WARN ][              kf.configwidgets] - A widget named ' "kcfg_AnalyzeAlternativeDirectoryName" ' was found but there is no setting named ' "AnalyzeAlternativeDirectoryName" '
[2024-10-03T16:51:23.163 CEST WARN ][                       default] - QObject::connect(QAction, Ekos::Scheduler): invalid nullptr parameter
[2024-10-03T16:51:23.208 CEST WARN ][                       default] - libpng warning: iCCP: known incorrect sRGB profile
[2024-10-03T16:51:23.246 CEST DEBG ][                org.kde.kstars] - Daylight Saving Time active
[2024-10-03T16:51:23.246 CEST DEBG ][                org.kde.kstars] - Next Daylight Savings Time change (Local Time):  "dim. oct. 27 01:00:00 2024 GMT"
[2024-10-03T16:51:23.246 CEST DEBG ][                org.kde.kstars] - Next Daylight Savings Time change (UTC):  "sam. oct. 26 23:00:00 2024 GMT"
[2024-10-03T16:51:23.315 CEST DEBG ][                org.kde.kstars] - Daylight Saving Time active
[2024-10-03T16:51:23.315 CEST DEBG ][                org.kde.kstars] - Next Daylight Savings Time change (Local Time):  "dim. oct. 27 01:00:00 2024 GMT"
[2024-10-03T16:51:23.315 CEST DEBG ][                org.kde.kstars] - Next Daylight Savings Time change (UTC):  "sam. oct. 26 23:00:00 2024 GMT"
[2024-10-03T16:51:23.315 CEST DEBG ][                org.kde.kstars] - Starting the timer
[2024-10-03T16:51:23.385 CEST DEBG ][                org.kde.kstars] - Date/Time is: "jeu. oct. 3 14:51:23 2024 GMT"
[2024-10-03T16:51:23.385 CEST DEBG ][                org.kde.kstars] - Location: "Paris, Paris, France"
[2024-10-03T16:51:23.385 CEST DEBG ][                org.kde.kstars] - TZ0: 1 TZ: 2
[2024-10-03T16:51:23.403 CEST WARN ][                       default] - QImage::scaled: Image is a null image
[2024-10-03T16:51:23.518 CEST WARN ][           org.kde.kstars.ekos] - Failed to automatically connect due to missing EkosLive credentials: "Entry not found"
[2024-10-03T16:51:24.050 CEST INFO ][                org.kde.kstars] - Loading comets
[2024-10-03T16:51:28.605 CEST INFO ][                org.kde.kstars] - Loading asteroids
[2024-10-03T16:59:41.330 CEST WARN ][                       default] - Skipped method "getDSSURL" : Type not registered with QtDBus in parameter list: float
[2024-10-03T16:59:41.330 CEST WARN ][                       default] - Skipped method "getDSSURL" : Type not registered with QtDBus in parameter list: float
[2024-10-03T16:59:41.330 CEST WARN ][                       default] - Skipped method "openFITS" : Type not registered with QtDBus in parameter list: QUrl
[2024-10-03T16:59:41.330 CEST WARN ][                       default] - Skipped method "slotShowPositionBar" : Pointers are not supported: SkyPoint*
[2024-10-03T16:59:41.330 CEST WARN ][                       default] - Skipped method "slotEyepieceView" : Pointers are not supported: SkyPoint*
[2024-10-03T16:59:41.330 CEST WARN ][                       default] - Skipped method "slotEyepieceView" : Pointers are not supported: SkyPoint*
[2024-10-03T16:59:41.330 CEST WARN ][                       default] - Skipped method "updateLocationFromWizard" : Unregistered input type in parameter list: GeoLocation
[2024-10-03T16:59:41.330 CEST WARN ][                       default] - Invalid return type in method "wiView"
[2024-10-03T16:59:41.331 CEST WARN ][                       default] - Unsupported return type 65 QPixmap in method "grab"
[... cut ...]
[2024-10-03T17:03:32.399 CEST DEBG ][           org.kde.kstars.indi] - Stopping INDI Server  "localhost" @ 7624
[2024-10-03T17:03:32.399 CEST DEBG ][           org.kde.kstars.indi] - INDI server disconnected. Exit code: -1
[2024-10-03T17:03:32.399 CEST DEBG ][           org.kde.kstars.ekos] - Resetting Ekos Manager...
[2024-10-03T17:03:32.400 CEST INFO ][           org.kde.kstars.ekos] - "Services INDI arrêtés."
[2024-10-03T17:03:32.413 CEST DEBG ][           org.kde.kstars.indi] - INDIListener: Removing client manager for server localhost @ 7624

closed 2024-12-10 by github-actions[bot]