indilib/indi #2124: Cannot compile latest 2.1.0

paolostivanin, 2024-10-01:
Hello,
I can't compile the latest release on any openSUSE codestream (15.5, 15.6 and Tumbleweed) due to:

```
[   38s] /home/abuild/rpmbuild/BUILD/indi-2.1.0/libs/indibase/stream/streammanager.cpp: In member function ‘bool INDI::StreamManagerPrivate::uploadStream(const uint8_t*, uint32_t)’:
[   38s] /home/abuild/rpmbuild/BUILD/indi-2.1.0/libs/indibase/stream/streammanager.cpp:1181:61: error: ‘class INDI::CCD’ has no member named ‘WebSocketS’; did you mean ‘WebSocketSP’?
[   38s]  1181 |                 && dynamic_cast<INDI::CCD*>(currentDevice)->WebSocketS[CCD::WEBSOCKET_ENABLED].s == ISS_ON)
[   38s]       |                                                             ^~~~~~~~~~
[   38s]       |                                                             WebSocketSP
[   38s] /home/abuild/rpmbuild/BUILD/indi-2.1.0/libs/indibase/stream/streammanager.cpp:1218:65: error: ‘class INDI::CCD’ has no member named ‘WebSocketS’; did you mean ‘WebSocketSP’?
[   38s]  1218 |                     && dynamic_cast<INDI::CCD*>(currentDevice)->WebSocketS[CCD::WEBSOCKET_ENABLED].s == ISS_ON)
[   38s]       |                                                                 ^~~~~~~~~~
[   38s]       |                                                                 WebSocketSP
[   38s] make[2]: *** [libs/indibase/CMakeFiles/indidriver_OBJECT.dir/build.make:93: libs/indibase/CMakeFiles/indidriver_OBJECT.dir/stream/streammanager.cpp.o] Error 1
[   38s] make[2]: Leaving directory '/home/abuild/rpmbuild/BUILD/indi-2.1.0/build'
[   38s] make[2]: *** Waiting for unfinished jobs....
```

darix, 2024-10-01:
```
[  9%] Building CXX object libs/indibase/CMakeFiles/indidriver_OBJECT.dir/indiccd.cpp.o
cd /home/abuild/rpmbuild/BUILD/indi-2.1.0/build/libs/indibase && /var/lib/build/ccache/bin/c++ -DBOOST_SYSTEM_DYN_LINK -DBOOST_SYSTEM_NO_LIB -DBOOST_THREAD_DYN_LINK -DBOOST_THREAD_NO_LIB -DENABLE_INDI_SHARED_MEMORY -DGIT_TAG_STRING=\"2.1.0-tgz\" -DHAVE_CLOCK_GETTIME -DHAVE_LIBNOVA -DHAVE_TIMESPEC_GET -DHAVE_WEBSOCKET -DWITH_ENCLEN -DWITH_MINMAX -I/home/abuild/rpmbuild/BUILD/indi-2.1.0/build/libs/indibase -I/home/abuild/rpmbuild/BUILD/indi-2.1.0/libs/indibase -I/home/abuild/rpmbuild/BUILD/indi-2.1.0/libs -I/home/abuild/rpmbuild/BUILD/indi-2.1.0/libs/httplib -I/usr/include/cfitsio -I/usr/include/libusb-1.0 -I/home/abuild/rpmbuild/BUILD/indi-2.1.0/libs/indibase/timer -I/home/abuild/rpmbuild/BUILD/indi-2.1.0/libs/indiclient -I/home/abuild/rpmbuild/BUILD/indi-2.1.0/libs/indiabstractclient -I/home/abuild/rpmbuild/BUILD/indi-2.1.0/libs/indicore -I/home/abuild/rpmbuild/BUILD/indi-2.1.0/libs/indidevice -I/home/abuild/rpmbuild/BUILD/indi-2.1.0/libs/indidevice/property -I/home/abuild/rpmbuild/BUILD/indi-2.1.0/build/libs/indicore -I/home/abuild/rpmbuild/BUILD/indi-2.1.0/libs/indibase/. -I/home/abuild/rpmbuild/BUILD/indi-2.1.0/libs/indibase/thread -I/home/abuild/rpmbuild/BUILD/indi-2.1.0/libs/indibase/stream -I/home/abuild/rpmbuild/BUILD/indi-2.1.0/libs/indibase/webcam -I/home/abuild/rpmbuild/BUILD/indi-2.1.0/libs/indicore/. -I/home/abuild/rpmbuild/BUILD/indi-2.1.0/build/libs/indicore/../.. -I/home/abuild/rpmbuild/BUILD/indi-2.1.0/libs/indidevice/. -I/home/abuild/rpmbuild/BUILD/indi-2.1.0/build/libs/indidevice/../.. -I/home/abuild/rpmbuild/BUILD/indi-2.1.0/libs/eventloop/. -I/home/abuild/rpmbuild/BUILD/indi-2.1.0/libs/dsp/. -I/home/abuild/rpmbuild/BUILD/indi-2.1.0/libs/fpack/. -I/home/abuild/rpmbuild/BUILD/indi-2.1.0/libs/hid/. -O2 -Wall -U_FORTIFY_SOURCE -D_FORTIFY_SOURCE=3 -fstack-protector-strong -funwind-tables -fasynchronous-unwind-tables -fstack-clash-protection -Werror=return-type  -g  -O1 -Wa,--noexecstack  -Wall -Wextra -Wno-format-truncation -g -DHAVE_MREMAP -O2 -g -DNDEBUG -std=gnu++17 -fPIC -MD -MT libs/indibase/CMakeFiles/indidriver_OBJECT.dir/indiccd.cpp.o -MF CMakeFiles/indidriver_OBJECT.dir/indiccd.cpp.o.d -o CMakeFiles/indidriver_OBJECT.dir/indiccd.cpp.o -c /home/abuild/rpmbuild/BUILD/indi-2.1.0/libs/indibase/indiccd.cpp
/home/abuild/rpmbuild/BUILD/indi-2.1.0/libs/indibase/indiccd.cpp: In member function 'virtual bool INDI::CCD::updateProperties()':
/home/abuild/rpmbuild/BUILD/indi-2.1.0/libs/indibase/indiccd.cpp:659:29: error: 'INDI::PropertyView<T>* INDI::PropertyBasic<T>::operator&() [with T = _ISwitch]' is protected within this context
  659 |             defineProperty(&WebSocketSP);
      |                             ^~~~~~~~~~~
In file included from /home/abuild/rpmbuild/BUILD/indi-2.1.0/libs/indidevice/property/indipropertyswitch.h:21,
                 from /home/abuild/rpmbuild/BUILD/indi-2.1.0/libs/indibase/indiccdchip.h:23,
                 from /home/abuild/rpmbuild/BUILD/indi-2.1.0/libs/indibase/indiccd.h:29,
                 from /home/abuild/rpmbuild/BUILD/indi-2.1.0/libs/indibase/indiccd.cpp:30:
/home/abuild/rpmbuild/BUILD/indi-2.1.0/libs/indidevice/property/indipropertybasic.h:122:27: note: declared protected here
  122 |         PropertyView<T> * operator &();
      |                           ^~~~~~~~
/home/abuild/rpmbuild/BUILD/indi-2.1.0/libs/indibase/indiccd.cpp:744:40: error: 'class INDI::PropertySwitch' has no member named 'name'
  744 |             deleteProperty(WebSocketSP.name);
      |                                        ^~~~
/home/abuild/rpmbuild/BUILD/indi-2.1.0/libs/indibase/indiccd.cpp:745:48: error: 'class INDI::PropertyNumber' has no member named 'name'
  745 |             deleteProperty(WebSocketSettingsNP.name);
      |                                                ^~~~
/home/abuild/rpmbuild/BUILD/indi-2.1.0/libs/indibase/indiccd.cpp: In member function 'virtual bool INDI::CCD::ISNewSwitch(const char*, const char*, ISState*, char**, int)':
/home/abuild/rpmbuild/BUILD/indi-2.1.0/libs/indibase/indiccd.cpp:1596:39: error: 'class INDI::PropertySwitch' has no member named 'name'
 1596 |         if (!strcmp(name, WebSocketSP.name))
      |                                       ^~~~
/home/abuild/rpmbuild/BUILD/indi-2.1.0/libs/indibase/indiccd.cpp:1598:29: error: 'INDI::PropertyView<T>* INDI::PropertyBasic<T>::operator&() [with T = _ISwitch]' is protected within this context
 1598 |             IUUpdateSwitch(&WebSocketSP, states, names, n);
      |                             ^~~~~~~~~~~
/home/abuild/rpmbuild/BUILD/indi-2.1.0/libs/indidevice/property/indipropertybasic.h:122:27: note: declared protected here
  122 |         PropertyView<T> * operator &();
      |                           ^~~~~~~~
/home/abuild/rpmbuild/BUILD/indi-2.1.0/libs/indibase/indiccd.cpp:1599:25: error: 'class INDI::PropertySwitch' has no member named 's'
 1599 |             WebSocketSP.s = IPS_OK;
      |                         ^
/home/abuild/rpmbuild/BUILD/indi-2.1.0/libs/indibase/indiccd.cpp:1601:17: error: 'WebSocketS' was not declared in this scope; did you mean 'WebSocketSP'?
 1601 |             if (WebSocketS[WEBSOCKET_ENABLED].s == ISS_ON)
      |                 ^~~~~~~~~~
      |                 WebSocketSP
/home/abuild/rpmbuild/BUILD/indi-2.1.0/libs/indibase/indiccd.cpp:1604:17: error: 'WebSocketSettingsN' was not declared in this scope; did you mean 'WebSocketSettingsNP'?
[... cut ...]
 2796 |             wsServer.send_binary(targetChip->FitsB.blob, targetChip->FitsB.bloblen);
      |                                                                      ^~~~~
      |                                                                      FitsBP
make[2]: *** [libs/indibase/CMakeFiles/indidriver_OBJECT.dir/build.make:429: libs/indibase/CMakeFiles/indidriver_OBJECT.dir/indiccd.cpp.o] Error 1
```

a few more places where it fails to build.

building against websocketpp-devel-0.8.2 on openSUSE Tumbleweed.

paolostivanin, 2024-10-02:
websocketpp 0.8.2 is the same being used on Ubuntu :thinking:

knro, 2024-10-02:
Should be fixed, we republished 2.1.0

closed 2024-10-02 by knro

darix, 2024-10-02:
Re-releasing something is a bad habit. some distros might already have fetched your tarball including its checksum and now get an alert about mismatches e.g.

The proper solution would have been to do 2.1.1.