indilib/indi #2127: indi_lx200_OnStep crashes "connection reset by peer"

hbwtrs, 2024-10-07:
Since the latest update to version 1.23 of the lx200_OnStep driver, the driver crashes immediately. The update was for kstars-bleeding and indi-full from Jasem's PPA, performed on October 4th.

Hardware/OS: 
1. Mele Quieter 2Q mini PC with Kubuntu 22.04 (also tested with a fresh install of Linux Mint XFCE 22 but the error remained)
2. Lenovo Ideapad 5 with Linux Mint 21.3

Mount:
1. Terrans Industries v4lite, running OnStep,
2. Warpastron WD-17 running OnStepX

The error only occurs when using the mini PC, not on the Lenovo laptop.

Solution was to downgrade the driver to version 1.22.

I checked the code changes from v1.22 to v1.23 in drivers/telescope/lx200_OnStep.cpp and drivers/telescope/lx200_OnStep.h. The code changes have to do with the connection to the mount, as does the error, but I cannot find the reason for this behavior.

Part of the logfile:
[2024-10-05T17:34:02.728 CEST DEBG ][           org.kde.kstars.indi] - LX200 OnStep : "[DEBUG] Toggle Debug Level -- Scope Verbose "
[2024-10-05T17:34:03.335 CEST INFO ][           org.kde.kstars.ekos] - Connecting to "LX200 OnStep"
[2024-10-05T17:34:03.336 CEST DEBG ][           org.kde.kstars.ekos] - Connecting  "LX200 OnStep"
[2024-10-05T17:34:03.358 CEST DEBG ][           org.kde.kstars.indi] - LX200 OnStep : "[DEBUG] Connecting to /dev/ttyUSB0 @ 9600 "
[2024-10-05T17:34:03.605 CEST DEBG ][           org.kde.kstars.indi] - LX200 OnStep : "[DEBUG] Port FD 3 "
[2024-10-05T17:34:03.605 CEST DEBG ][           org.kde.kstars.indi] - LX200 OnStep : "[DEBUG] Connection successful, attempting handshake... "
[2024-10-05T17:34:03.621 CEST DEBG ][           org.kde.kstars.indi] - LX200 OnStep : "[DEBUG] Testing telescope connection using ACK... "
[2024-10-05T17:34:12.481 CEST INFO ][           org.kde.kstars.ekos] - "Failed to connect to LX200 OnStep. Please ensure device is connected and powered on."
[2024-10-05T17:34:13.466 CEST DEBG ][           org.kde.kstars.indi] - LX200 OnStep : "[DEBUG] Failure. Telescope is not responding to ACK! "
[2024-10-05T17:34:13.466 CEST DEBG ][           org.kde.kstars.indi] - LX200 OnStep : "[SCOPE] CMD <:GVP#> "
[2024-10-05T17:34:13.488 CEST DEBG ][           org.kde.kstars.indi] - LX200 OnStep : "[SCOPE] RES <0> "
[2024-10-05T17:34:13.488 CEST DEBG ][           org.kde.kstars.indi] - LX200 OnStep : "[SCOPE] CMD <:GVP#> "
[2024-10-05T17:34:13.517 CEST DEBG ][           org.kde.kstars.indi] - LX200 OnStep : "[SCOPE] RES <O> "
[2024-10-05T17:34:13.517 CEST DEBG ][           org.kde.kstars.indi] - LX200 OnStep : "[DEBUG] Testing telescope connection using ACK... "
[2024-10-05T17:34:13.532 CEST DEBG ][           org.kde.kstars.indi] - LX200 OnStep : "[DEBUG] Testing successful! "
[2024-10-05T17:34:13.534 CEST INFO ][           org.kde.kstars.indi] - LX200 OnStep :  "[INFO] LX200 OnStep is online. "
[2024-10-05T17:34:13.534 CEST INFO ][           org.kde.kstars.ekos] - "LX200 OnStep" Version: "1.23" Interface: 141 is connected.
.....
.....
....
[2024-10-05T17:34:15.654 CEST DEBG ][           org.kde.kstars.indi] - INDI Server:  "2024-10-05T15:34:15: Driver indi_lx200_OnStep: read: Connection reset by peer"
[2024-10-05T17:34:15.654 CEST DEBG ][           org.kde.kstars.indi] - INDI Server:  "<delProperty device=\"LX200 OnStep\"/>"
[2024-10-05T17:34:15.654 CEST DEBG ][           org.kde.kstars.indi] - INDI Server:  "2024-10-05T15:34:15: Driver indi_lx200_OnStep: Terminated after #0 restarts."
[2024-10-05T17:34:15.654 CEST DEBG ][           org.kde.kstars.indi] - Removing device "LX200 OnStep"
[2024-10-05T17:34:15.654 CEST DEBG ][           org.kde.kstars.indi] - INDI Server:  ""
[2024-10-05T17:34:15.654 CEST CRIT ][           org.kde.kstars.indi] - INDI driver  "indi_lx200_OnStep"  crashed!
[2024-10-05T17:34:15.685 CEST DEBG ][           org.kde.kstars.indi] - INDIListener: Removing device "LX200 OnStep"
[2024-10-05T17:34:15.686 CEST DEBG ][     org.kde.kstars.ekos.mount] - Removing mount driver "LX200 OnStep"
[2024-10-05T17:34:15.693 CEST INFO ][           org.kde.kstars.ekos] - "LX200 OnStep is offline."

knro, 2024-10-07:
Please use this script to generate the logs:

```
#!/bin/bash

filename=~/crash_$(date +%Y%m%d_%H%M%S).txt
echo "Starting INDI driver debugging for" $*
gdb -batch -ex "handle SIG33 nostop noprint" -ex "set follow-fork-mode child" -ex "run" -ex "bt" --args indiserver -r 0 -v $* > $filename 2>&1
echo "Debugging finished. Please send" $filename "file to support."
```

Save it as igdb.sh and copy it to /usr/bin and then chmod +x to make it executable.

Then run it as this:

`igdb indi_lx200OnStep`

If it crashes, send the logs and attach it here.

hbwtrs, 2024-10-07:
Jasem,

as hopefull as I was before, I'm completely puzzled now. I've had cases where the script does not result in an error, but the same driver v1.23 started from Ekos always crashes. I also had one case where the v1.22 driver crashed when started from the script, with the same error, but never when starting from Ekos. 

Tested again on two systems, two mounts. I attached two files, with all detailed information written in the header. Even though I documented all my steps methodically, I get the odd strange result now and then. It bothers me when thereś no logical pattern in the results 100% of the time.

Main takeaway is that the script gives the same error as the Ekos logs with output (two files attached with more info):

[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
2024-10-07T18:45:08: startup: /usr/bin/indiserver -r 0 -v /usr/bin/indi_lx200_OnStep
[Attaching after Thread 0x7ffff7f90740 (LWP 5530) fork to child process 5538]
[New inferior 2 (process 5538)]
2024-10-07T18:45:08: Driver /usr/bin/indi_lx200_OnStep: pid=5538 rfd=6 wfd=6 efd=7
2024-10-07T18:45:08: listening to port 7624 on fd 5
2024-10-07T18:45:08: Local server: listening on local domain at: @/tmp/indiserver
[Detaching after fork from parent process 5530]
[Inferior 1 (process 5530) detached]
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
process 5538 is executing new program: /usr/bin/indi_lx200generic
2024-10-07T18:45:08: Driver /usr/bin/indi_lx200_OnStep: /usr/bin/indi_lx200_OnStep: symbol lookup error: /usr/bin/indi_lx200_OnStep: undefined symbol: _ZN4INDI15GuiderInterfaceC2Ev
2024-10-07T18:45:08: Driver /usr/bin/indi_lx200_OnStep: read: Connection reset by peer
2024-10-07T18:45:08: Driver /usr/bin/indi_lx200_OnStep: Terminated after #0 restarts.
2024-10-07T18:45:08: good bye
[Inferior 2 (process 5538) exited with code 0177]
No stack.

[crash_MeleQuieter2Q_Kubuntu22.04_KStars3.7.2_lx200_OnstepX_v1.22.txt](https://github.com/user-attachments/files/17283642/crash_MeleQuieter2Q_Kubuntu22.04_KStars3.7.2_lx200_OnstepX_v1.22.txt)
[crash_MeleQuieter2Q_Kubuntu22.04_KStars3.7.2_lx200_OnstepX_v1.23.txt](https://github.com/user-attachments/files/17283643/crash_MeleQuieter2Q_Kubuntu22.04_KStars3.7.2_lx200_OnstepX_v1.23.txt)

hbwtrs, 2024-10-07:
So 15 minutes ago I had my Ekos working like before, using v1.22 of the driver. I shutdown the computer, put the whole telescope setup back together. put it on the mount ready to go outside, same error again. Maybe it's not the indi_lx200_OnStep version, but then what is it?

hbwtrs, 2024-10-07:
As hopeful as I was 45 minutes ago, I'm now back where I started. I had a workaround by downgrading the driver, mounted everything on the telescope, put it on the mount, all ready. Now I cannot get it to work at all, I get "Connection reset by peer" on both my mounts, with different versions of the driver. If only I could understand the logic behind this behavior.

knro, 2024-10-08:
Looks like something in the PPA build didn't build correctly. INDI was updated in the PPA about 9 hours ago, can you try updating again?

hbwtrs, 2024-10-08:
Just tested the new version. The driver still crashes, but with a different output than last time, so I think we have a small step towards succes.

- PC: Mele Quieter 2Q
- OS: Linux 6.8.0-45-generic 22.04.1-Ubuntu SMP PREEMPT_DYNAMIC
- Mount: WarpAstron WD-17 running OnStepX
- KStars/Indi new install using: apt remove kstars-bleeding indi-full; apt autoremove; apt update; apt upgrade; apt install kstars-bleeding indi-full; reboot
- Configuration files not removed before new install
- Version:
      - kstars-bleeding/jammy,now 6:3.7.3+202410080857~ubuntu22.04.1 amd64 [installed]
     - indi-full/jammy,now 2.1.0+202410011955~ubuntu22.04.1 amd64 [installed]
- Result from KStars: crash with different result than on October 7th (kstars logfile attached)
- Result from debug script: no crash (killed script using CTRL-C) (output debug script attached)

[LX200_log20-53-41.txt](https://github.com/user-attachments/files/17298263/LX200_log20-53-41.txt)
[crash_20241008_210728.txt](https://github.com/user-attachments/files/17298264/crash_20241008_210728.txt)

knro, 2024-10-09:
It's fixed and verified in master. Please check if you can from source. We will rebuild INDI later tonight.

closed 2024-10-09 by knro

hbwtrs, 2024-10-09:
> It's fixed and verified in master. Please check if you can from source. We will rebuild INDI later tonight.

Thanks! I checked on two PCs, all seems to work fine now.

Now there's just one tiny annoying thing with OnStepX, not really sure if I should make a bug report on feature request of this new issue. I'll first try and find it in the code myself.