indilib/indi #2128: Telescope Simulator custom track rate only updates when track mode is changed

PeterS-mBryonics, 2024-10-07:
**Describe the bug**
Using the telescope simulator, updating the custom tracking rates only occurs when the track mode is changed (for example custom -> sidereal -> custom), rather than on set of new tracking rates

**To Reproduce**
1. Run driver indi_simulator_telescope
2. Set track mode to "Custom"
3. Change track rates ra/dec to 5000, 5000
4. Set track rates
5. (No change in tracking rate)
6. Change Track Mode to "Sidereal" and then back to "Custom"
7. Observe motion of telescope

**Expected behavior**
When a custom track rate is set, the rate is immediately applied

**Desktop (please complete the following information):**
 - OS: WSL Ubuntu 22.04
 - Version: Release 2.0.9 and latest git

**Log Files**
(also can't add log files because the driver turns debug off after it's turned on)

knro, 2024-10-08:
This is already fixed in master. Can you please check now?

PeterS-mBryonics, 2024-10-08:
Thanks, cleaned out my copy and re-fetched the main branch, and recompiled etc. It has the same behaviour as before

closed 2024-10-08 by knro

knro, 2024-10-08:
This was not enough.. during the migration to INDI new style properties a regression was introduced. Hopefully it's fixed now, please recompile from master and try

PeterS-mBryonics, 2024-10-08:
Fab, that got it! Cheers.

commit 68b456e8 (2024-10-08): Fix setting custom track rates. Fixes #2128