indilib/indi #2132: iexos-100 pmc-8 mount has wrong tracking rates

NeTox1c, 2024-10-08:
**Describe the bug**
I have a iexos-100 pmc-8 mount. So far everything is working with it withing EKOS, but the tracking rate is totally off. 
When I set it to sidereal it sets 15.00000 instead of the correct 15.041067. 
The option for setting a custom rate does not work. 
With an older Ekos/Indi (pmc8 driver version 0.4) the tracking rate was correct and setting custom rates was also working.

**To Reproduce**
Exact steps to reproduce the behavior.
1. Connect to an iexos-100 pmc-8 mount through Ekos/Indi
2. start tracking
3a. choose sidereal -> bug
3b. try setting a custom rate -> bug

**Expected behavior**
The tracking rate of "sidereal" should be set to 15.041067 and not 15.000000.
When setting a custom tracking rate, the value should be send to the mount and the mount should track with it.

**Screenshots**
In the added screenshot you can see in the log below, that I am switching between Lunar, Solar and Sidereal. None of the tracking-rates are correct.
[image]

**Desktop (please complete the following information):**
 - OS: Raspberry Pi 4 OS (different versions tested), Stellarmate, Astroberry-Image (mount is working with this old image)

**Log Files**
Make sure to enable logging and include log files [log_08-43-48.txt](https://github.com/user-attachments/files/17289102/log_08-43-48.txt)

jfwells, 2024-12-03:
There has been a bit of back and forth about this in the past, with the tracking rate changing to 15.041 as requested by another user previously. However, this is not correct, and caused incorrect tracking and reporting on my iExos-100 mount. According to the docs, the tracking rate is communicated using "arc-seconds per sidereal second", which is 15.000..

This is confirmed in the documentation here: https://02d3287.netsolhost.com/pmc-eight/PMC_Eight_ProgrammersReferenceManual_Release2_2019_February_01.pdf

and was confirmed here: https://espmc-eight.groups.io/g/MAIN/topic/96696552?p=Created,,,20,2,0,0

Every time this gets changed back to 15.041 it is a regression.

closed 2024-12-03 by knro

knro, 2024-12-03:
Thanks, I submitted changes to the code. Just to confirm, I changed all the rates to the following according to the table in the PDF file:
```
#define PMC8_RATE_SIDEREAL 15.000
#define PMC8_RATE_LUNAR 14.451
#define PMC8_RATE_SOLAR 14.959
#define PMC8_RATE_KING 14.996
```

However,  since INDI uses tracking rates in arcsecs per *solar* second, I added a conversion factor in the code now. Can you please double check the code and see if it works OK? From the user perspective, the tracking rates would match what everyone expects similar to ASCOM..etc. Internally, it converts to PMC8 units.