indilib/indi #2143: IEXOS100 - incorrect handling of goto in southern hemisphere for ASIAIR

GreendaleObs, 2024-10-31:
**Describe the bug**
after alignment - goto's in southern hemisphere do line up 

https://bbs.zwoastro.com/d/15720-pmc8-exos2

When I start the ASIair Plus it indicates my correct location on the mount, but the Skymap continues to point to the north celestial pole.

When executing the Home Position, the time in AR changes from 16 hours to 4 hours (12 hours difference, but without moving the engine) and in DEC from +90° it goes to -89°, completing a turn of approximately 180°. Just there, the skymap indicates that the mount is pointing at the South Celestial Pole.

When I indicate to go to an object (Canopus star for example), the engine moves with inverted direction (when the location on the mount is in my position and recognize that i'm on the southern hemisphere) For example, canpus at this time is passing trough the meridian to the west, so the counterweight has to pint to the west, BUT NO, goes to the east and the DEC goes to tie oposite direction.

The INDI driver of the mount indicates my time in UTC -5, but in my location I am in UTC -3.

User Maxi Falieres and I can test in south for you 

Textme to my mail [maxijfalieres@gmail.com](mailto:maxijfalieres@gmail.com) and maybe we can chat to get solution to this.

ASIAIR contact is Evan Wu - developer for ASIAIR

GreendaleObs, 2024-11-22:
Can we get an update here - no one looked at this ???

knro, 2024-11-23:
We don't know what version of INDI AIRAir is using so it's hard to debug. Can you confirm if you have the same issue with StellarMate (which has latest INDI) or if you want to recompile INDI yourself.

closed 2025-01-31 by github-actions[bot]