indilib/indi #2145: Jobs scheduled to run at midnight, are scheduled for midnight of the preceding day

stevecoh1, 2024-11-01:
**Describe the bug**
When a job is scheduled with a date of midnight the next day, it appears on the schedule as midnight the current day.  If the user fails to notice this, and starts the schedule, the job executes immediately, rather than at the intended start time.  The workaround is to start the job at 00:01 rather than midnight.

**To Reproduce**
Very simple to reproduce:
Enter a job with a start time of midnight, the following day.
When the job is added to the schedule, it appears as midnight of the current day (i.e. in the past)

**Expected behavior**
The job should be entered in the schedule as it was entered by the user, i.e. midnight tomorrow.

**Screenshots**

**Desktop (please complete the following information):**
Stellarmate Pro, although I reproduced it using the VNC interface, i.e. on the device itself.

**Log Files**

stevecoh1, 2024-11-01:
screenshots included
1. User entered tomorrow at midnight but not added to schedule
[image]
2. After user clicked (+) to add to the schedule. 
[image]

stevecoh1, 2025-01-01:
Uh, why has no one looked at it?

knro, 2025-01-04:
Because this is a KStars issue, not INDI.

KStars: Ekos and planetariums
INDI: Drivers only

closed 2025-01-04 by knro

stevecoh1, 2025-01-04:
But **is** this only a KStars issue?  I'm pretty sure I first noticed it on Stellarmate app, and only showed it here in KStars because I didn't want the INDI folks to think it was a Stellarmate issue.  In any case, I easily replicated it this morning as shown in these screenshots, on the Stellarmate App.

[image]
[image]

I suppose it's possible that both the Stellarmate App and KStars have the same bug.

I did submit a similar but not identical issue on the Stellarmate site in which you said 
"Thank you for the report. For any issues in KStars that are not StellarMate specific, I highly recommend you post it on INDI forums where KStars developers can take a look."

I think it should be reopened and looked at.

stevecoh1, 2025-01-04:
> scheduled for midnight of the preceding day (Issue #2145)
>
> Closed #2145 <https://github.com/indilib/indi/issues/2145> as completed.
>
> <https://github.com/indilib/indi/issues/2145#event-15814033867>, or
> unsubscribe

I commented on the Github site that I can replicate the same problem on the
Stellarmate App.  Please have a look.

---------- Forwarded message ---------