indilib/indi #2148: Just purchased new ZWO AM5N Mount.  INDI won't connect to it.

stevecoh1, 2024-11-12:
**Describe the bug**
I just purchased a brand new ZWO AM5N (2024 version) mount.  My Stellarmate Pro won't connect to it.

**To Reproduce**
Exact steps to reproduce the behavior.
1. Obtain mount and connect it to Stellarmate Pro using the USB port
2. Create a Profile using the AM5 USB driver 
3. Try to start it.  Connection fails. Log complains that /dev/ttyUSB0 does not exist, which is indeed the case.  Config file generated for me by INDI mentions /dev/ttyUSB0.  

**Expected behavior**
I expected a connection to be made

**Screenshots**
Screenshot showing config file and lsusb output

**Desktop (please complete the following information):**
 - OS: Stellarmate 1.8.5

 - Version 1.8.5

**Log Files**
Log was sent to Stellarmate, Issue see #1000

knro, 2024-11-12:
Scan ports and it should be detected. Use the Port Selector tool. I literally connected to a new AMN5 yesterday with SM Pro without any issues.

stevecoh1, 2024-11-12:
Thanks, Jasem.
I'm unfamiliar with that tool since I've only used Celestron AVX up until now.
I guess you mean [this](https://www.stellarmate.com/help/PortSelector.html).
I also see [this, making use of the Bluetooth capabilities of the mount](https://www.stellarmate.com/help/SerialPortAssistantBluetoothWiza.html), which sounds intriguing because of one less cable, maybe.  
Would one be better than the other?

knro, 2024-11-12:
Yes, port selector. I think bluetooth is not working reliably, we're working on improving that.

stevecoh1, 2024-11-12:
It worked.  That was easy enough. thanks.

closed 2024-11-12 by stevecoh1