indilib/indi #2156: libindi.pc doesn't provide any -l in Libs

DarthGandalf, 2024-12-24:
**Describe the bug**
The purpose of pkg-config is to tell build system of the client of the library what flags to pass to compiler in order to use the library. However, libindi.pc's Libs simply says `Libs: -L${libdir}`, resulting in undefined symbols in linker stage, unless the client also explicitly provides `-lindiclient` or such.

**To Reproduce**
Exact steps to reproduce the behavior.
1. Run `pkg-config --libs libindi`
2. See empty output

**Expected behavior**
It should return the list of libraries to link to.

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Desktop (please complete the following information):**
 - OS: Gentoo Linux
 - Version 2.1.1

In https://github.com/indilib/indi/blob/master/libindi.pc.cmake the libraries are supposed to come from `@PKG_CONFIG_LIBS@`, but cmake never sets this variable

DarthGandalf, 2025-02-22:
comment for unstaleing

knro, 2025-02-22:
Do you a PR to fix this issue?

DarthGandalf, 2025-02-22:
I don't know what the correct fix should be. I only use libindiclient.so, but I also see libindidriver.so, libindilx200.so and a few more. I don't know how these other .so files are supposed to be used and even what are they.

If those files are internal to indi, it would be enough to simply replace `@PKG_CONFIG_LIBS@` with `-lindiclient` in libindi.pc

If those files are supposed to be linked to by other programs in some cases, probably it makes sense to have separate .pc files for indiclient and some combination of other libraries provided: e.g. have libindi-client.pc vs libindi-AlignmentDriver.pc

DarthGandalf, 2025-04-24:
Well, let's attempt #2230