indilib/indi #2167: [ONSTEP] Crash when sync, 100% of the times

arakeis, 2025-01-31:
crash with onstep lx200 driver when trying to sync

**To Reproduce**
1. Run driver onstep lx200
2. unpark, slew not far from nsp, resolution with sync
3. The driver INDI LX200 OnStep has crashed. Restart ? (loosely translated from french)

**Expected behavior**
un clean sync beetween the computed position and the onstep driven mount

**Desktop (please complete the following information):**
 - OS: StellarmateOS
 - Version 1.8.6 

**Log Files**
[2025-01-31T10:22:03.407 CET INFO ][     org.kde.kstars.ekos.align] - "Résolution terminée après 0.31 secondes."
[2025-01-31T10:22:03.408 CET INFO ][     org.kde.kstars.ekos.align] - "Résolveur AD (2.47168) DEC (89.85886) Orientation (90.13485) Échelle de pixels (2.97977) Parité (neg)"
[2025-01-31T10:22:03.409 CET INFO ][     org.kde.kstars.ekos.align] - "Coordonnées de la solution : AD (11h 27m 51s) DEC ( 89° 59' 48\") Coordonnées du télescope : AD (00h 11m 46s) DEC ( 90° 00' 00\") Coordonnées de la cible : AD (00h 11m 41s) DEC ( 90° 00' 00\")"
[2025-01-31T10:22:03.411 CET DEBG ][     org.kde.kstars.ekos.align] - Align state changed to "Succès"
[2025-01-31T10:22:03.413 CET DEBG ][     org.kde.kstars.ekos.align] - Align state changed to "Synchronisation"
[2025-01-31T10:22:03.413 CET DEBG ][           org.kde.kstars.indi] - ISD:Telescope: Syncing...
[2025-01-31T10:22:03.415 CET DEBG ][           org.kde.kstars.indi] - ISD:Telescope sending coords RA: "11h 27m 51s" ( 11.4642 ) DE: " 89° 59' 48\"" ( 89.9968 )
[2025-01-31T10:22:03.417 CET INFO ][     org.kde.kstars.ekos.align] - "Synchronisation aux coordonnées : AD (11h 27m 51s) DEC ( 89° 59' 48\")"
[2025-01-31T10:22:03.458 CET DEBG ][           org.kde.kstars.indi] - INDI Server:  "2025-01-31T09:22:03: Driver indi_lx200_OnStep: *** stack smashing detected ***: terminated"
[2025-01-31T10:22:03.458 CET DEBG ][           org.kde.kstars.indi] - INDI Server:  ""
[2025-01-31T10:22:03.460 CET DEBG ][           org.kde.kstars.indi] - INDI Server:  "2025-01-31T09:22:03: Driver indi_lx200_OnStep: read EOF"
[2025-01-31T10:22:03.460 CET DEBG ][           org.kde.kstars.indi] - INDI Server:  "<delProperty device=\"LX200 OnStep\"/>"
[2025-01-31T10:22:03.460 CET DEBG ][           org.kde.kstars.indi] - INDI Server:  "2025-01-31T09:22:03: Driver indi_lx200_OnStep: Terminated after #0 restarts."
[2025-01-31T10:22:03.460 CET DEBG ][           org.kde.kstars.indi] - INDI Server:  ""
[2025-01-31T10:22:03.460 CET CRIT ][           org.kde.kstars.indi] - INDI driver  "indi_lx200_OnStep"  crashed!
[2025-01-31T10:22:03.499 CET DEBG ][           org.kde.kstars.indi] - Removing device "LX200 OnStep"
[2025-01-31T10:22:03.592 CET INFO ][           org.kde.kstars.indi] - CCD Simulator :  "[INFO] CCD FOV rotation updated to 90.1348 degrees. "
[2025-01-31T10:22:03.593 CET INFO ][           org.kde.kstars.indi] - LX200 OnStep :  "[ERROR] Unexpected return on sync call! "
[2025-01-31T10:22:03.594 CET INFO ][           org.kde.kstars.indi] - LX200 OnStep :  "[ERROR] Check system & Align if doing align to see if it went through! "
[2025-01-31T10:22:03.597 CET DEBG ][           org.kde.kstars.indi] - INDIListener: Removing device "LX200 OnStep"
[2025-01-31T10:22:03.597 CET DEBG ][     org.kde.kstars.ekos.mount] - Removing mount driver "LX200 OnStep"
[2025-01-31T10:22:03.601 CET INFO ][           org.kde.kstars.ekos] - "Le périphérique LX200 OnStep est déconnecté."
[2025-01-31T10:22:03.672 CET INFO ][ org.kde.kstars.ekos.scheduler] - ""
[2025-01-31T10:22:05.761 CET DEBG ][           org.kde.kstars.indi] - restartDriver with no cm, and  2  drivers. Trying to remove:  "LX200 OnStep"
[2025-01-31T10:22:06.762 CET DEBG ][           org.kde.kstars.indi] - Adding managed driver "LX200 OnStep"
[2025-01-31T10:22:06.762 CET DEBG ][           org.kde.kstars.indi] - Starting INDI Driver  "indi_lx200_OnStep"
[2025-01-31T10:22:06.762 CET DEBG ][           org.kde.kstars.indi] - INDI Server:  "2025-01-31T09:22:06: FIFO: start indi_lx200_OnStep"
[2025-01-31T10:22:06.762 CET DEBG ][           org.kde.kstars.indi] - INDI Server:  "2025-01-31T09:22:06: FIFO: Starting driver indi_lx200_OnStep"
[2025-01-31T10:22:06.762 CET DEBG ][           org.kde.kstars.indi] - INDI Server:  "2025-01-31T09:22:06: Driver indi_lx200_OnStep: pid=2927 rfd=10 wfd=10 efd=15"
[2025-01-31T10:22:06.762 CET DEBG ][           org.kde.kstars.indi] - INDI Server:  ""
[2025-01-31T10:22:06.774 CET DEBG ][           org.kde.kstars.indi] - INDI Server:  "2025-01-31T09:22:06: Driver indi_lx200_OnStep: initializing from LX200 OnStep device..."
[2025-01-31T10:22:06.775 CET DEBG ][           org.kde.kstars.indi] - INDI Server:  "2025-01-31T09:22:06: Driver indi_lx200_OnStep: Initializing from Generic LX200 device..."
[2025-01-31T10:22:06.775 CET DEBG ][           org.kde.kstars.indi] - INDI Server:  ""
[2025-01-31T10:22:06.777 CET DEBG ][           org.kde.kstars.indi] - INDI Server:  "2025-01-31T09:22:06: Driver indi_lx200_OnStep: snooping on Dome Simulator.DOME_PARK"
[2025-01-31T10:22:06.777 CET DEBG ][           org.kde.kstars.indi] - INDI Server:  "2025-01-31T09:22:06: Driver indi_lx200_OnStep: snooping on Dome Simulator.DOME_SHUTTER"
[2025-01-31T10:22:06.777 CET DEBG ][           org.kde.kstars.indi] - INDI Server:  ""
[2025-01-31T10:22:06.778 CET DEBG ][           org.kde.kstars.indi] - Received new device LX200 OnStep

knro, 2025-01-31:
Please debug via [Ekos Debugger](https://stellarmate.com/support/tutorials/ekos-debugger.html) and submit the crash backtrace.

arakeis, 2025-01-31:
ok, will give it a try

arakeis, 2025-01-31:
here is the first lines of the ekos debugging (i have installed the ekos debugger dbsym package, yet the log is quite unreadable ...

[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/aarch64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffebff7c00 (LWP 3670)]
[New Thread 0x7fffe97e7c00 (LWP 3671)]
[New Thread 0x7fffe6fd7c00 (LWP 3672)]
[New Thread 0x7fffe3fbfc00 (LWP 3673)]
[New Thread 0x7fffe27ffc00 (LWP 3674)]
[Thread 0x7fffe27ffc00 (LWP 3674) exited]
[New Thread 0x7fffe27ffc00 (LWP 3675)]
[New Thread 0x7fffe1fefc00 (LWP 3676)]
[Thread 0x7fffe27ffc00 (LWP 3675) exited]
[Thread 0x7fffe1fefc00 (LWP 3676) exited]
[New Thread 0x7fffe1fefc00 (LWP 3677)]
[New Thread 0x7fffe27ffc00 (LWP 3678)]
[Thread 0x7fffe27ffc00 (LWP 3678) exited]
[Thread 0x7fffe1fefc00 (LWP 3677) exited]
[New Thread 0x7fffe27ffc00 (LWP 3679)]
[New Thread 0x7fffe1fefc00 (LWP 3680)]
[Thread 0x7fffe27ffc00 (LWP 3679) exited]
[Thread 0x7fffe1fefc00 (LWP 3680) exited]
[New Thread 0x7fffe1fefc00 (LWP 3681)]
[New Thread 0x7fffd3fefc00 (LWP 3682)]
[New Thread 0x7fffd1fdfc00 (LWP 3683)]
[New Thread 0x7fffcffcfc00 (LWP 3684)]
[New Thread 0x7fffc7fefc00 (LWP 3685)]
[New Thread 0x7fffe27ffc00 (LWP 3686)]
[New Thread 0x7fffb14dfc00 (LWP 3687)]
[New Thread 0x7fffb0ccfc00 (LWP 3688)]
[New Thread 0x7fffabfefc00 (LWP 3689)]
[New Thread 0x7fffab137c00 (LWP 3690)]
[New Thread 0x7fffaa927c00 (LWP 3691)]
[New Thread 0x7fffaa117c00 (LWP 3692)]
[New Thread 0x7fffa9907c00 (LWP 3693)]
[New Thread 0x7fffa90f7c00 (LWP 3694)]
[New Thread 0x7fff8ffefc00 (LWP 3695)]
[New Thread 0x7fff8f7dfc00 (LWP 3696)]
[Thread 0x7fffab137c00 (LWP 3690) exited]
[Thread 0x7fffa9907c00 (LWP 3693) exited]
[Detaching after fork from child process 3697]
[Detaching after fork from child process 3698]
[Detaching after fork from child process 3699]
[New Thread 0x7fffa9907c00 (LWP 3702)]
[New Thread 0x7fffab137c00 (LWP 3706)]
[New Thread 0x7fff8e47bc00 (LWP 3707)]
[Thread 0x7fffaa117c00 (LWP 3692) exited]
[Thread 0x7fffa90f7c00 (LWP 3694) exited]
[Thread 0x7fffaa927c00 (LWP 3691) exited]

arakeis, 2025-01-31:
here is the two logs files from ekos debuger (indi tab)

[app.indi.debug.log](https://github.com/user-attachments/files/18617923/app.indi.debug.log)
[indi.debug.log](https://github.com/user-attachments/files/18617922/indi.debug.log)

knro, 2025-01-31:
```
2025-01-31T11:59:11: Driver indi_lx200_OnStep: *** stack smashing detected ***: terminated
```
So it crashes here but unfortunately, all I can tell is that it happens somewhere in ISNewNumber but nothing more specific. Can you recompile INDI from Github and test again?

arakeis, 2025-01-31:
here they are

stellarmate@stellarmate:~/indi-3rdparty $ indiserver -v
2025-01-31T14:47:56: startup: indiserver -v
Usage: indiserver [options] driver [driver ...]
Purpose: server for local and remote INDI drivers
INDI Library: 2.1.1
Code v2.1.1-32-gd3473444a. Protocol 1.7.

[app.indi.debug.log](https://github.com/user-attachments/files/18620166/app.indi.debug.log)
[indi.debug.log](https://github.com/user-attachments/files/18620167/indi.debug.log)

arakeis, 2025-02-20:
Hi Jason, any inspiration looking at the debug file?

Could it be that sync with onstep works only while in onstep own alignment mode?

knro, 2025-02-22:
Sorry for delay, from the logs:

```
Using host libthread_db library "/lib/aarch64-linux-gnu/libthread_db.so.1".
Thread 2.1 "indi_lx200_OnSt" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7ffff76da280 (LWP 11629)]
0x00742d746573202c in ?? ()
#0  0x00742d746573202c in ?? ()
#1  0x61742d746573202c in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
```

So usually these types of crashes are hard to diagnose, let me check if I can reproduce the issue.

knro, 2025-02-22:
Ok at least from local testing, I cannot reproduce the crash:

```
2025-02-22T05:53:28: [INFO] OnStep: Synchronization successful. 
2025-02-22T05:53:00: [INFO] Slewing to RA: 15:20:45 - DEC: 71:44:12 
2025-02-22T05:51:32: [INFO] Initialization Complete 
2025-02-22T05:51:32: [INFO] Mount is unparked. 
```

arakeis, 2025-02-22:
Great news, sort off, just to be clear are you testing with an actual onstep mount or are you in simulation mode? I'll test in that mode and report, I'll test with or without the on-board GPS of my onstep board.

knro, 2025-02-22:
I just tested using OnStepX on Arduino board. No physical motors.

arakeis, 2025-02-22:
I think I found the culprit of the bug.
My mount has a built-in GPS managed by OnStep.
It works quite well, BUT on the first boot or when used indoors, the mount is in an in-between state where no GPS position is set.
After placing my mount outside, which is quite risky given the current weather in France, the GPS position is acquired, and the sync works.
I'm further investigating to confirm the gps path to the potential bug.

arakeis, 2025-04-04:
I thought I found the problem in the config file of ONSTEP
(PIER_SIDE_SYNC_CHANGE_SIDES)
but no...
reinstalled my stellarmate setup (hardware change), every thing working
now, go figure...

closed 2025-04-04 by knro