indilib/indi #2174: Compilation fails with Boost 1.87.0

paolostivanin, 2025-02-28:
Hello,
latest INDI fails to build with Boost 1.87.0 due to:

```
[   30s] cd /home/abuild/rpmbuild/BUILD/indi-2.1.2.1-build/indi-2.1.2.1/build/libs/indibase && /usr/bin/c++ -DBOOST_SYSTEM_DYN_LINK -DBOOST_SYSTEM_NO_LIB -DBOOST_THREAD_DYN_LINK -DBOOST_THREAD_NO_LIB -DENABLE_INDI_SHARED_MEMORY -DGIT_TAG_STRING=\"2.1.2-tgz\" -DHAVE_CLOCK_GETTIME -DHAVE_LIBNOVA -DHAVE_TIMESPEC_GET -DHAVE_WEBSOCKET -DHAVE_XISF -DWITH_ENCLEN -I/home/abuild/rpmbuild/BUILD/indi-2.1.2.1-build/indi-2.1.2.1/build/libs/indibase -I/home/abuild/rpmbuild/BUILD/indi-2.1.2.1-build/indi-2.1.2.1/libs/indibase -I/home/abuild/rpmbuild/BUILD/indi-2.1.2.1-build/indi-2.1.2.1/libs -I/home/abuild/rpmbuild/BUILD/indi-2.1.2.1-build/indi-2.1.2.1/libs/httplib -I/usr/include/cfitsio -I/usr/include/libusb-1.0 -I/home/abuild/rpmbuild/BUILD/indi-2.1.2.1-build/indi-2.1.2.1/libs/indibase/timer -I/home/abuild/rpmbuild/BUILD/indi-2.1.2.1-build/indi-2.1.2.1/libs/indiclient -I/home/abuild/rpmbuild/BUILD/indi-2.1.2.1-build/indi-2.1.2.1/libs/indiabstractclient -I/home/abuild/rpmbuild/BUILD/indi-2.1.2.1-build/indi-2.1.2.1/libs/indicore -I/home/abuild/rpmbuild/BUILD/indi-2.1.2.1-build/indi-2.1.2.1/libs/indidevice -I/home/abuild/rpmbuild/BUILD/indi-2.1.2.1-build/indi-2.1.2.1/libs/indidevice/property -I/home/abuild/rpmbuild/BUILD/indi-2.1.2.1-build/indi-2.1.2.1/build/libs/indicore -I/home/abuild/rpmbuild/BUILD/indi-2.1.2.1-build/indi-2.1.2.1/libs/indibase/. -I/home/abuild/rpmbuild/BUILD/indi-2.1.2.1-build/indi-2.1.2.1/libs/indibase/thread -I/home/abuild/rpmbuild/BUILD/indi-2.1.2.1-build/indi-2.1.2.1/libs/indibase/stream -I/home/abuild/rpmbuild/BUILD/indi-2.1.2.1-build/indi-2.1.2.1/libs/indibase/webcam -I/home/abuild/rpmbuild/BUILD/indi-2.1.2.1-build/indi-2.1.2.1/libs/indicore/. -I/home/abuild/rpmbuild/BUILD/indi-2.1.2.1-build/indi-2.1.2.1/build/libs/indicore/../.. -I/home/abuild/rpmbuild/BUILD/indi-2.1.2.1-build/indi-2.1.2.1/libs/indidevice/. -I/home/abuild/rpmbuild/BUILD/indi-2.1.2.1-build/indi-2.1.2.1/build/libs/indidevice/../.. -I/home/abuild/rpmbuild/BUILD/indi-2.1.2.1-build/indi-2.1.2.1/libs/eventloop/. -I/home/abuild/rpmbuild/BUILD/indi-2.1.2.1-build/indi-2.1.2.1/libs/dsp/. -I/home/abuild/rpmbuild/BUILD/indi-2.1.2.1-build/indi-2.1.2.1/libs/fpack/. -I/home/abuild/rpmbuild/BUILD/indi-2.1.2.1-build/indi-2.1.2.1/libs/hid/. -O2 -Wall -U_FORTIFY_SOURCE -D_FORTIFY_SOURCE=3 -fstack-protector-strong -funwind-tables -fasynchronous-unwind-tables -fstack-clash-protection -Werror=return-type -flto=auto -ffat-lto-objects -g -Wno-stringop-truncation  -O1 -Wa,--noexecstack  -Wall -Wextra -Wno-format-truncation -g -DHAVE_MREMAP -O2 -g -DNDEBUG -std=gnu++17 -fPIC -MD -MT libs/indibase/CMakeFiles/indidriver_OBJECT.dir/stream/streammanager.cpp.o -MF CMakeFiles/indidriver_OBJECT.dir/stream/streammanager.cpp.o.d -o CMakeFiles/indidriver_OBJECT.dir/stream/streammanager.cpp.o -c /home/abuild/rpmbuild/BUILD/indi-2.1.2.1-build/indi-2.1.2.1/libs/indibase/stream/streammanager.cpp
[   30s] In file included from /usr/include/websocketpp/transport/asio/endpoint.hpp:32,
[   30s]                  from /usr/include/websocketpp/config/asio_no_tls.hpp:32,
[   30s]                  from /home/abuild/rpmbuild/BUILD/indi-2.1.2.1-build/indi-2.1.2.1/libs/indibase/indiwsserver.h:22,
[   30s]                  from /home/abuild/rpmbuild/BUILD/indi-2.1.2.1-build/indi-2.1.2.1/libs/indibase/indisensorinterface.h:28,
[   30s]                  from /home/abuild/rpmbuild/BUILD/indi-2.1.2.1-build/indi-2.1.2.1/libs/indibase/stream/streammanager_p.h:39,
[   30s]                  from /home/abuild/rpmbuild/BUILD/indi-2.1.2.1-build/indi-2.1.2.1/libs/indibase/stream/streammanager.cpp:26:
[   30s] /usr/include/websocketpp/transport/asio/connection.hpp:89:24: error: ‘io_service’ in namespace ‘websocketpp::lib::asio’ does not name a type
[   30s]    89 |     typedef lib::asio::io_service * io_service_ptr;
[   30s]       |                        ^~~~~~~~~~
[   30s] /usr/include/websocketpp/transport/asio/connection.hpp:91:40: error: ‘io_service’ is not a member of ‘websocketpp::lib::asio’
[   30s]    91 |     typedef lib::shared_ptr<lib::asio::io_service::strand> strand_ptr;
[   30s]       |                                        ^~~~~~~~~~
[   30s] /usr/include/websocketpp/transport/asio/connection.hpp:91:58: error: template argument 1 is invalid
[   30s]    91 |     typedef lib::shared_ptr<lib::asio::io_service::strand> strand_ptr;
[   30s]       |                                                          ^
[   30s] /usr/include/websocketpp/transport/asio/connection.hpp:461:32: error: ‘io_service_ptr’ has not been declared
[   30s]   461 |     lib::error_code init_asio (io_service_ptr io_service) {
[   30s]       |                                ^~~~~~~~~~~~~~
[   30s] /usr/include/websocketpp/transport/asio/connection.hpp:1175:5: error: ‘io_service_ptr’ does not name a type
[   30s]  1175 |     io_service_ptr  m_io_service;
[   30s]       |     ^~~~~~~~~~~~~~
[   30s] /usr/include/websocketpp/transport/asio/connection.hpp: In member function ‘websocketpp::transport::asio::connection<config>::timer_ptr websocketpp::transport::asio::connection<config>::set_timer(long int, websocketpp::transport::timer_handler)’:
[   30s] /usr/include/websocketpp/transport/asio/connection.hpp:316:18: error: ‘m_io_service’ was not declared in this scope; did you mean ‘m_is_server’?
[   30s]   316 |                 *m_io_service,
[   30s]       |                  ^~~~~~~~~~~~
[   30s]       |                  m_is_server
[   30s] /usr/include/websocketpp/transport/asio/connection.hpp:321:43: error: base operand of ‘->’ is not a pointer
[   30s]   321 |             new_timer->async_wait(m_strand->wrap(lib::bind(
[   30s]       |                                           ^~
[   30s] /usr/include/websocketpp/transport/asio/connection.hpp: In member function ‘std::error_code websocketpp::transport::asio::connection<config>::init_asio(int)’:
[   30s] /usr/include/websocketpp/transport/asio/connection.hpp:462:9: error: ‘m_io_service’ was not declared in this scope; did you mean ‘io_service’?
[   30s]   462 |         m_io_service = io_service;
[   30s]       |         ^~~~~~~~~~~~
[   30s]       |         io_service
[   30s] /usr/include/websocketpp/transport/asio/connection.hpp:465:22: error: request for member ‘reset’ in ‘((websocketpp::transport::asio::connection<config>*)this)->websocketpp::transport::asio::connection<config>::m_strand’, which is of non-class type ‘websocketpp::transport::asio::connection<config>::strand_ptr’ {aka ‘int’}
[   30s]   465 |             m_strand.reset(new lib::asio::io_service::strand(*io_service));
[   30s]       |                      ^~~~~
[   30s] /usr/include/websocketpp/transport/asio/connection.hpp:465:43: error: ‘io_service’ in namespace ‘websocketpp::lib::asio’ does not name a type
[   30s]   465 |             m_strand.reset(new lib::asio::io_service::strand(*io_service));
[   30s]       |                                           ^~~~~~~~~~
[   30s] /usr/include/websocketpp/transport/asio/connection.hpp: In member function ‘void websocketpp::transport::asio::connection<config>::handle_post_init(timer_ptr, websocketpp::transport::init_handler, const std::error_code&)’:
[   30s] /usr/include/websocketpp/transport/asio/connection.hpp:576:58: error: ‘using std::__shared_ptr_access<boost::asio::basic_waitable_timer<std::chrono::_V2::steady_clock>, __gnu_cxx::_S_atomic, false, false>::element_type = class boost::asio::basic_waitable_timer<std::chrono::_V2::steady_clock>’ {aka ‘class boost::asio::basic_waitable_timer<std::chrono::_V2::steady_clock>’} has no member named ‘expires_from_now’
[   30s]   576 |             (post_timer && lib::asio::is_neg(post_timer->expires_from_now())))
[   30s]       |                                                          ^~~~~~~~~~~~~~~~
[   30s] /usr/include/websocketpp/transport/asio/connection.hpp: In member function ‘void websocketpp::transport::asio::connection<config>::proxy_write(websocketpp::transport::init_handler)’:
[   30s] /usr/include/websocketpp/transport/asio/connection.hpp:632:25: error: base operand of ‘->’ is not a pointer
[   30s]   632 |                 m_strand->wrap(lib::bind(
[   30s]       |                         ^~
[   30s] /usr/include/websocketpp/transport/asio/connection.hpp: In member function ‘void websocketpp::transport::asio::connection<config>::proxy_read(websocketpp::transport::init_handler)’:
[... cut ...]
[   30s]       |                               ~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~
[   30s] make[2]: *** [libs/indibase/CMakeFiles/indidriver_OBJECT.dir/build.make:96: libs/indibase/CMakeFiles/indidriver_OBJECT.dir/stream/streammanager.cpp.o] Error 1
[   30s] make[2]: Leaving directory '/home/abuild/rpmbuild/BUILD/indi-2.1.2.1-build/indi-2.1.2.1/build'
[   30s] make[1]: *** [CMakeFiles/Makefile2:2758: libs/indibase/CMakeFiles/indidriver_OBJECT.dir/all] Error 2
[   30s] make[1]: *** Waiting for unfinished jobs....
```

SloppyJaconda, 2025-03-11:
Boost ASIO removed `asio::io_service` in [Boost 1.87.0](https://github.com/boostorg/asio/blob/032545fc569553ff9a1f142dd9cec1a9e070a4f9/doc/history.qbk#L108). Per ASIO's documentation, it was an alias to `asio::io_context`. 

I did some quick looking around and it looks like that reference is being pulled in via `websocketpp`. The original `websocketpp` doesn't seem like it's been getting any attention from the owner for the last 5 years, however there is a [PR](https://github.com/zaphoyd/websocketpp/pull/1164) to address that and a couple forks linked within that have merged it.

knro, 2025-03-11:
I removed Websocket support, it wasn't really used and the need for it dropped when we supported FAST Blobs.

closed 2025-03-11 by knro