indilib/indi #2180: Indi Drivers Crashed

cropduster32, 2025-03-02:
Over the last few nights observing an error message"

"Indi drivers: indi_asi_ccd crashed", restart in 10 sec

appears and aborts both guiding and image capture>
The cameras are a Zwo ASI585MC and a Zwo ASI120MM

I cannot reproduce the error as it appears randomly either immediately after setting guiding on the run or part way through image capture

Happy to supply any logs that may be required but haven't a clue where to find them.

The system I'm using is an RPi4 running Stellarmate and networked to a laptop running Ubuntu 22.04. The RPi4 is connected via a VNC server over a WiFi network where the router is close to the RPi4.

knro, 2025-03-11:
Any backtraces? Use Ekos Debugger to get the backtrace.

knro, 2025-04-07:
Most likely timeout leading to driver restart. These are known to happen with 120

closed 2025-04-07 by knro