indilib/indi #2194: Libindi deconfliction of peripherals

maphilli14, 2025-03-13:
I'm unable to post for several attempts over several weeks in the forum.

I have a mixed use RaspberryPi4 for homeassistant / zigbee (use case for those interested - https://www.youtube.com/watch?v=-UPKvhtHlFk )
The ports used for zigbee control are something like:

[code] HubZ Smart Home Controller - /dev/ttyUSB1, s/n: 51501996 - Silicon Labs - 10C4:8A2A From astroberry@astrocart:~ $ lsusb Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub Bus 001 Device 003: ID 10c4:8a2a Cygnal Integrated Products, Inc. Bus 001 Device 004: ID 06cd:0121 Keyspan USA-19hs serial adapter Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub[/code]

Anytime I connect devices to astroberry in device manager the zigbee plug gets usurped and goes offline.  How can exclude something like

[code]/dev/ttyUSB 0 and 1 as 2 is the mount, with other devices pending... ls /dev/ttyU* /dev/ttyUSB0  /dev/ttyUSB1  /dev/ttyUSB2 [/code]

TallFurryMan, 2025-03-13:
You should write a udev rule to create a custom device symlink with an explicit name for your configuration. You cannot rely on the generic device names to be stable.

TallFurryMan, 2025-03-13:
A few examples here : https://github.com/TallFurryMan/astro-slave/tree/master/udev, although your mileage may vary.

maphilli14, 2025-03-14:
I'm fine with an OS level filter or security mechanism, though I do not fully understand them yet.  Is there nothing within AstroBerry or docker/homeassistant that can reserve the hardware so I can't be hijacked?

maphilli14, 2025-03-14:
The zigbee/z-wave stick is pretty static, here's the list of devices with no other indi devices inserted yet.

:/Docker/homeassistant# ls /dev/ttyU*
/dev/ttyUSB0  /dev/ttyUSB1

TallFurryMan, 2025-03-14:
I have no strong experience with astroberry unfortunately, but I believe your problem is more at the OS than the application level. Devices are enumerated by udev and the kernel. When a new device appears it will normally be added to the /dev set non-destructively, that is, without touching other devices, unless a USB hub reset is introduced. However, if your homeassistant runs inside docker, there may be an additional mapping that implies a side effect, that I don't see yet. I admit I didn't check your video tough, there may be an architecture diagram there? I'll get back after watching.

TallFurryMan, 2025-03-14:
@maphilli14 your setup is simply awesome.

TallFurryMan, 2025-03-14:
I found https://stackoverflow.com/a/66427245 to be a good reference on what you can try to map a precisely identified device in your homeassistant container. That is, if I properly understood your problem.

knro, 2025-03-14:
Use `/dev/serial/by-id`

TallFurryMan, 2025-03-14:
> Use `/dev/serial/by-id`

Indeed. Seems I can't wrap my head to use modern kernels!

maphilli14, 2025-03-15:
Thanks all, JFYI, indi seems fine to run in the background, it's only when the drivers are invoked from a client does the homeassistant go offline/unavailable.  See screenshot

[image]

Solution https://github.com/indilib/indi/issues/2194#issuecomment-2723844118  seems very complicated! 😐

maphilli14, 2025-03-15:
Here's why - 

[image]

I feel like it would be easier to get astroberry/indilib to just NOT probe or regester or request resources from /dev/X, /dev/Y is there really no means to put an exclusion list together?

maphilli14, 2025-03-15:
Sorry for the slow rolling debug here, but it seem that once indilib gets its hooks into the usb used by HA it won't let go with a restart of HA docker.  I had a full restart of the RasbperrryPi to resolve as I don't know how astroberry controls the INDI under the hood.  A simple stop of services didn't release the lock.

[image]

maphilli14, 2025-03-15:
Here's the gui indi setup, the highlighted one is where the Mount is connected.

[image]

However it throws an error

[image]

I'm happy to just turn autodiscovery off and configure manually if possible, but is there a baud rate or parity setting I'm missing that is causing the errors?

maphilli14, 2025-03-15:
I'm not convinced it works at all regardless

```
2025-03-15T21:19:41: [ERROR] Read Command Error: Timeout error 
2025-03-15T21:19:36: [ERROR] Read Command Error: Timeout error 
2025-03-15T21:19:31: [ERROR] Read Command Error: Timeout error 
2025-03-15T21:19:26: [INFO] Mount is unparked. 
2025-03-15T21:19:26: [INFO] InitPark: No Park data in file /home/astroberry/.indi/ParkData.xml: No park data found for this device 
2025-03-15T21:19:26: [ERROR] Read Command Error: Timeout error 
2025-03-15T21:19:21: [ERROR] Read Command Error: Timeout error 
2025-03-15T21:19:16: [ERROR] Read Command Error: Timeout error 
2025-03-15T21:19:11: [ERROR] Read Command Error: Timeout error 
2025-03-15T21:19:06: [ERROR] Read Command Error: Timeout error 
2025-03-15T21:19:01: [ERROR] Read Command Error: Timeout error 
2025-03-15T21:18:47: [INFO] Device configuration saved. 
2025-03-15T21:18:47: [INFO] Saving device configuration... 
2025-03-15T21:18:47: [INFO] iOptron CEM120 is online. 
2025-03-15T21:18:47: [INFO] Trying connecting to /dev/serial/by-id/usb-Silicon_Labs_HubZ_Smart_Home_Controller_51501996-if01-port0 @ 230400 ... 
2025-03-15T21:18:45: [WARNING] Communication with /dev/serial/by-id/usb-Keyspan__a_division_of_InnoSys_Inc._Keyspan_USA-19H-if00-port0 @ 230400 failed. Starting Auto Search... 
2025-03-15T21:18:45: [ERROR] Read Command Error: Timeout error 
2025-03-15T21:18:42: [ERROR] Read Command Error: Timeout error 
2025-03-15T21:18:31: [INFO] Auto search is enabled. When connecting, the driver shall attempt to communicate with all available system ports until a connection is established. 
```

TallFurryMan, 2025-03-15:
Some drivers connect without verifying the hardware they talk to is really the expected one. It seem iOptron is one of those.

If you turned off auto search and saved configuration, I wonder why your log is starting with this sentence. In that situation, it simply appears only your zigbee is accepting the connection, which I agree is unfortunate, then the driver saves that and messes everything up. 

The iOptron firmware is communicating at 115200 bauds, use that as speed, select the serial port manually, disable autosearch, then save your configuration (and ensure it was properly saved).

maphilli14, 2025-03-16:
Well @TallFurryMan  that seemed to do the trick, thanks!!!

[image]

I'll fool around a bit more and revert back to you!  Thanks everyone!