indilib/indi #2207: indi_qsi_ccd buffer overflow on amd64 arch

astromick71, 2025-03-23:
Description of Bug
Hi, I've been having a few problems with my QSI cameras when trying to use them with Kstars/Ekos/libindi and my Ubuntu laptop (amd64). When trying to capture images with QSI 583wsg/683wsg-8 cameras using the indi_qsi_ccd driver there is a buffer overflow then the driver crashes on the amd64 arch.

Symptoms
Start up Kstars/Ekos (or indiserver indi_qsi_ccd) and connect to QSI 583wsg/683wsg-8. At this point the camera is connected and shutter/filter wheel/temperature control all appear functional. During capture the shutter opens and closes (can be heard and seen). During the image download phase a "buffer overflow error" is reported. This happens every time Under Ubuntu 24.04.2 LTS using indi-full and latest version of kstars-bleeding on the amd64 arch. However, the same cameras and cables work fine using the same setup (including cabling) on the arm64 arch and also work fine using the latest QSI application on a Windows 11 laptop. 

 - OS: Ubuntu 24.04.2 LTS (amd64)
 - Git indi-full and kstars-bleeding (latest)

Please let me know if you need additional information and logs.

Many thanks,

Mick

astromick71, 2025-05-03:
Hi,

Sorry to report that I am seeing this problem on the rpi5 (arm64) after upgrading from Ubuntu 24.04.1 LTS to 24.04.2 LTS. So it appears to be related to changes in Ubuntu and is not architecture specific.

Many thanks,

Mick

cgobat, 2025-10-30:
I am seeing the same issue on both Linux Mint 22.2 (which is based on Ubuntu 24.04) on an 64-bit Intel laptop and on Ubuntu 24.04 on an RPi 3B (aarch64). Camera works fine with Windows/ASCOM. I will try Arch and Fedora next to see if they are any different.

knro, 2025-10-31:
Install Rustdesk on the Ubuntu 24 x86_64 system and then send me ID/Password over email so I can access it remotely and debug.

cgobat, 2025-10-31:
The x86 Ubuntu 24.04 system was @astromick71's.

cgobat, 2025-11-11:
I loaded a coredump with gdb and was able to print this backtrace with `bt`. Not sure if it's helpful.

[indi_qsi_ccd_coredump_bt.txt](https://github.com/user-attachments/files/23486571/indi_qsi_ccd_coredump_bt.txt)

This is from my Linux Mint amd64 system.

cgobat, 2026-01-11:
Were you able to glean anything from that backtrace? I'm still running into this crash on a 64-bit x86 machine that I just did a brand-new fresh install of Linux Mint (22.2) on. I still have the coredump file that I can extract information from if there's anything else that would help with debugging.

cgobat, 2026-02-03:
@knro were you able to glean anything from that backtrace? I'm still running into this crash on a 64-bit x86 machine that I just did a brand-new install of Linux Mint (22.2) on. I can grab the core dump if there's anything else that would help with debugging.

svalat, 2026-03-10:
On my side I also seen this bug on ubuntu 24.04 with QSI 683-wsg since a moment.

It is probably related so something with the glibc or another user-space library because I succeed to make the camera working again by using another distribution in a docker container (older debian or gentoo).

By moving the host install to Gentoo (up to date), I don't see anymore the problem.

Note : the command tools coming with the QSI driver to detect the camera also crash if is is faster to reproduce the issue than clicking into EKOS (qsiapidemo or qsiapitest) I forgot which one or the two.