indilib/indi #2208: indi_gpio driver not working. Connection reset by peer

RhoLuke, 2025-03-23:
**Describe the bug**
Try to work with `indi_gpio` module from remote. The driver is working on a Rpi 4.

**To Reproduce**
Exact steps to reproduce the behavior.
1. Run driver `indiserver indi_gpio`
2. run Kstars/Ekos form the client and start the remote connection
3. See error `Driver indi_gpio: read: Connection reset by peer`

**Expected behavior**
It should work as describe in the documentation page of the driver

**Screenshots**

[image]

**Desktop (please complete the following information):**
Distributor ID:	Ubuntu
Description:	Ubuntu 24.04.2 LTS
Release:	24.04
Codename:	noble

knro, 2025-03-25:
Run it via Ekos Debugger or gdb and let's see where it crashes.

RhoLuke, 2025-03-27:
First of all, it's an honour to met you Jasem :)

Second, I'm trying to install the ekos degubber but I cannot because apt doesn't let me to add the nightly ppa repo.
Here the log

[image]

knro, 2025-03-28:
You don't need nightly, you can install the stable PPA.

RhoLuke, 2025-03-28:
Yes I already have also the PPA stable repo but it seems that it cannot find the `ekos-debugger`.

`Error: Unable to locate package ekos-debugger`

Does it changed the name maybe? 
I found the `sudo apt-get -y install ekos-debugger` from your thread in the indilib forum

RhoLuke, 2025-03-29:
Because I'm trying to running the `indi_gpio` from a remote RPi, I saw from the technical doc that in order to running on remote, the best option is to set and ssh connection in this way
`ssh -L 7624:localhost:7624 user@remote_host`

So I try the following procedure:
- establish the ssh connection: `ssh -L 7624:localhost:7624 pi4obs@pi4obs`
- start the indiserver on the RPi: `indiserver indi_simulator_telescope inidi_simulator_ccd indi_gpio`
- open `Kstars` the `Ekos`
- run a remote indiserver from `localhost`

But still don't connect GPIO driver, even if now from the RPi temrinal I don't get the _connection reset by the peer_ message.
Here the Kstar's log:

```
2025-03-29T18:42:37 INDI services stopped.
2025-03-29T18:42:16 Unable to remotely establish:
+ GPIO
Please ensure the device is connected and powered on.
2025-03-29T18:42:06 CCD Simulator filter is online.
2025-03-29T18:42:06 Guider port from CCD Simulator is ready.
2025-03-29T18:42:06 CCD Simulator is online.
2025-03-29T18:42:06 Guider port from Telescope Simulator is ready.
2025-03-29T18:42:06 Telescope Simulator is online.
2025-03-29T18:42:05 INDI services started. Connection to remote INDI server localhost:7,624 is successful. Waiting for devices...
2025-03-29T18:42:05 Connecting to remote INDI server at localhost on port 7,624 ...
```

knro, 2025-03-30:
You don't need to do that, just use INDI Web Manager on the remote system to start/stop INDI drivers. Maybe it's an Ubuntu vs Debian issue as well (not sure). Compiling from source might help.

RhoLuke, 2025-03-30:
> Another issue is permission, can you use gpioinfo, gpioset, gpioget without running sudo?

Yes I fix this problem and now the gipod functions are working without sudo. But the problem still.

> Compiling from source might help.

In order to not mess up everything, could you guide me a bit? So I think I have to purge the current indilib installed by apt, clone the indilib repo and build from there. Is it correct? There is something that I should be aware of? (some CMake configuration or something else..)

RhoLuke, 2025-03-30:
I tried also with `ekos-debugger` but it doesn't give any additional info.

That's an example of log form INDI tab:
```
2025-03-30T09:56:39: startup: /usr/bin/indiserver -r 0 -v indi_gpio indi_simulator_ccd indi_simulator_telescope
2025-03-30T09:56:39: Driver indi_gpio: pid=25537 rfd=6 wfd=6 efd=7
2025-03-30T09:56:39: Driver indi_simulator_ccd: pid=25538 rfd=8 wfd=8 efd=9
2025-03-30T09:56:39: Driver indi_simulator_telescope: pid=25539 rfd=10 wfd=10 efd=11
2025-03-30T09:56:39: listening to port 7624 on fd 5
2025-03-30T09:56:39: Local server: listening on local domain at: @/tmp/indiserver
2025-03-30T09:56:39: Driver indi_simulator_telescope: HaAxis: TrackRate 1, trackingRateDegSec 15.041067 arcsec
2025-03-30T09:56:39: Driver indi_simulator_ccd: snooping on Telescope Simulator.EQUATORIAL_EOD_COORD
2025-03-30T09:56:39: Driver indi_simulator_ccd: snooping on Telescope Simulator.EQUATORIAL_COORD
2025-03-30T09:56:39: Driver indi_simulator_ccd: snooping on Telescope Simulator.TELESCOPE_INFO
2025-03-30T09:56:39: Driver indi_simulator_ccd: snooping on Telescope Simulator.GEOGRAPHIC_COORD
2025-03-30T09:56:39: Driver indi_simulator_ccd: snooping on Telescope Simulator.TELESCOPE_PIER_SIDE
2025-03-30T09:56:39: Driver indi_simulator_ccd: snooping on CCD Simulator.FILTER_SLOT
2025-03-30T09:56:39: Driver indi_simulator_ccd: snooping on CCD Simulator.FILTER_NAME
2025-03-30T09:56:39: Driver indi_simulator_ccd: snooping on SQM.SKY_QUALITY
2025-03-30T09:56:39: Driver indi_simulator_telescope: snooping on Dome Simulator.DOME_PARK
2025-03-30T09:56:39: Driver indi_simulator_telescope: snooping on Dome Simulator.DOME_SHUTTER
2025-03-30T10:02:37: Driver indi_gpio: read EOF
<delProperty device="GPIO"/>
2025-03-30T10:02:37: Driver indi_gpio: Terminated after #0 restarts.
```

This from KStars tab:
```
This GDB supports auto-downloading debuginfo from the following URLs:
  <https://debuginfod.ubuntu.com>
Enable debuginfod for this session? (y or [n]) [answered N; input not from terminal]
Debuginfod has been disabled.
To make this setting permanent, add 'set debuginfod enabled off' to .gdbinit.
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffeefff6c0 (LWP 25732)]
[New Thread 0x7fffed7466c0 (LWP 25734)]
[New Thread 0x7fffecf456c0 (LWP 25735)]
[New Thread 0x7fffdffff6c0 (LWP 25736)]
[New Thread 0x7fffdf7fe6c0 (LWP 25737)]
[New Thread 0x7fffdeffd6c0 (LWP 25738)]
[New Thread 0x7fffde7fc6c0 (LWP 25739)]
[New Thread 0x7fffddffb6c0 (LWP 25740)]
[New Thread 0x7fffc7fff6c0 (LWP 25741)]
[New Thread 0x7fffc5ffe6c0 (LWP 25742)]
[New Thread 0x7fffc3ffd6c0 (LWP 25743)]
[New Thread 0x7fffb3fff6c0 (LWP 25744)]
[New Thread 0x7fffa9dff6c0 (LWP 25745)]
[New Thread 0x7fffabfff6c0 (LWP 25746)]
[New Thread 0x7fffa7dfe6c0 (LWP 25747)]
[New Thread 0x7fff97fff6c0 (LWP 25748)]
[New Thread 0x7fffdd4b66c0 (LWP 25749)]
[New Thread 0x7fffdccb56c0 (LWP 25750)]
[New Thread 0x7fffc11ff6c0 (LWP 25751)]
[New Thread 0x7fffb19986c0 (LWP 25752)]
[New Thread 0x7fffb11976c0 (LWP 25753)]
[New Thread 0x7fffb09966c0 (LWP 25754)]
[New Thread 0x7fffa5dfd6c0 (LWP 25755)]
[Thread 0x7fffb19986c0 (LWP 25752) exited]
[Thread 0x7fffb09966c0 (LWP 25754) exited]
[New Thread 0x7fffb09966c0 (LWP 25798)]
[Thread 0x7fffdffff6c0 (LWP 25736) exited]
[Thread 0x7fffa5dfd6c0 (LWP 25755) exited]
[Thread 0x7fffb11976c0 (LWP 25753) exited]
```

Maybe I'm not using it correctly.

RhoLuke, 2025-03-30:
Ok, I change OS to Raspbian 12 (bookworm), build the indi-core library and added the `indi_gpio` and now it works!

Now I have to understand how to set some pins as output in order to use this driver as a relays board driver

closed 2025-06-06 by github-actions[bot]