indilib/indi #2211: Question about creating web client for API

Astro-Otter-Space, 2025-03-25:
Hello,
I would like to create a web client for my own use (and for fun ^^) using a JS framework (nuxt or next...i don't know yet, but with a backend part). I saw this [tutorial](https://docs.indilib.org/clients/tutorial.html) but i don't know programming in C/C++. 
Before other question: is it possible ? I guess the client side language doesn't matter, as long as it can communicate with the INDI server.
How do i send query and parameters to API ? Not like curl query ? Do i need using websocket ?
For device, should i send drive name ? (for example, i run server like `indiserver indi_asi_ccd indi_libcamera`. How i connect these devices from client ?

I have only work with SOAP or Rest API, so it's a bit new ^^.
Thank you for answers :).

TallFurryMan, 2025-03-25:
Your best bet is probably to find or build bindings, so that you may develop in your language of choice. Pyindi-client provides bindings for Python, for instance. 

The other way is to communicate directly using XML messages, of course, through the socket provided by indiserver.

knro, 2025-03-26:
I'm adding INDI Alpaca bridge (still in progress), so maybe after it's done you can use one of the Alpaca tools and communicate with the drivers this way?

closed 2025-06-02 by github-actions[bot]