indilib/indi #2212: ZWO EAF Driver crash after update

vancurad, 2025-04-01:
**Describe the bug**

I just got a Unihedron SQM-LU delivered and wanted to try the connection with Kstars (but that's besides the point and unrelated I think).
While already being connected to Kstars and not planning on imaging, I figured I could update the software, so did a quick apt update & upgrade which did seem to update Indi drivers.

Primary issue: After starting Kstars again I'm now having issues connecting to my ZWO EAF and the driver appears to crash immediately after connecting to the device.
Secondary issue: I started seeing "Failed to get perc power information (ASI_INVALID_CONTROL_TYPE)" on my guide camera as well. I found [this post](https://indilib.org/forum/general/14676-solved-camera-errors-after-last-upgrade.html#100761) about similar logs. From the UI it appears that Ekos now believes my guide camera is cooled.

Detail: 
PC: Raspberry Pi 5, running Ubuntu Server
Camera: ASI2600MM Pro
Guide Camera: ASI174MM
Filter Wheel: ZWO EFW
Focuser: ZWO EAF
Mount: ZWO AM5N
Sky Meter: Unihedron SQM-LU

All devices were connected.

**To Reproduce**
Exact steps to reproduce the behavior.
1. Start Kstars
2. Open Ekos
3. Start Ekos

**Expected behavior**
ZWO EAF connects and works

**What is happening**
ZWO EAF appears to connect but then the driver crashes and it disappears.

**Screenshots**

[image]

[image]

**Desktop (please complete the following information):**
 - OS: Ubuntu 24.10
 - Version 24.10

**Log Files**

[eaf_crash.txt](https://github.com/user-attachments/files/19545951/eaf_crash.txt)

[apport.txt](https://github.com/user-attachments/files/19545956/apport.txt)

I'm not sure how to get or install Ekos Debugger

knro, 2025-04-01:
I see the builds failed in the PPA. Triggering a rebuild now.

knro, 2025-04-01:
Looks like like this will take more time as the INDI-3rdparty repo on Launchpad is out of date and the import is not working. I [url=https://answers.launchpad.net/launchpad/+question/821378]asked a question on Launchapd[/url] and hopefully they would fix this soon.

vancurad, 2025-04-01:
Thanks for working on it, really appreciate your commitment and effort on this project! 
I'm looking at a few cloudy nights coming up here anyways, so I for my part am not missing out on anything if my gear doesn't work right now. 😉

vancurad, 2025-04-02:
@knro Not sure if the issue you suspect would also be expected to reproduce when running the source code off master, but I compiled Indi today per instructions in the readme and just gave it a test run (not sure if my steps here were quite right):
1. Start indiserver verbose and connect only my EAF (`indiserver -vvv Indi_asi_focuser`)
2. Remote-connect from my laptop (latest stable release 3.7.6 for MacOS)
3. Start Ekos
4. Same locally on my Raspberry Pi 5: build & run Kstars straight from repo (you don't want to know how long the compile time on that little thing is, hah), launch it as usual and still seeing a crash with my focuser

Possible that there's something that still needs fixing that I'm not aware of and isn't solved with what I can get from the repositories of Kstars & Indi. In case it helps though, attached is the verbose output of indiserver when I connected from my laptop. Not 100% sure what to look for there, but if there is any unknown issue at all, this part here looks like it might be of interest:

```
2025-04-02T07:28:53: Client 9: queuing <setNumberVector device='' name=''>
2025-04-02T07:28:53: Client 9: sending msg nq 1:
<setNumberVector device="" name="" state="Idle" timeout="0" timestamp="2025-04-02T07:28:53"/>

2025-04-02T07:28:53: Driver indi_asi_focuser: 1 ZWO EAF attached out of 1 detected.Impossible ISState -144766176
2025-04-02T07:28:53: Driver indi_asi_focuser: read: Connection reset by peer
<delProperty device="ZWO EAF"/>
2025-04-02T07:28:53: Client 9: queuing <delProperty device='ZWO EAF' name=''>
2025-04-02T07:28:53: Driver indi_asi_focuser: restart #0
[Detaching after fork from child process 31685]
2025-04-02T07:28:53: Driver indi_asi_focuser: pid=31685 rfd=7 wfd=7 efd=10
2025-04-02T07:28:53: Driver indi_asi_focuser: sending msg nq 1:
<getProperties version="1.7"/>
```

[gdb.txt](https://github.com/user-attachments/files/19563807/gdb.txt)

I might try to revert to older release tags tomorrow.

vancurad, 2025-04-02:
Side note: I just checked out version 2.1.2.1 and I can successfully connect through steps 1-3 in my previous comment. All is working well on this version 🙂

knro, 2025-04-07:
It should be working now with latest GIT & Launchpad.

closed 2025-04-07 by knro

vancurad, 2025-04-15:
Confirmed, seems to be working fine now, thanks! :)