indilib/indi #2213: Can't access forums at indilib.org

Jimbolaya64, 2025-04-01:
This may be the wrong place for this but i can't access the indilib forums after creating and activating a new account. I log in just fine. I click the link to get to the forums and I get this error: "Cannot save session: User 0 doesn't exist!". Other links along the top menu seem to work fine. I apologize for posting here but I obviously can't access the forums to post the problem, the discord link at the bottom of the home page does not work and I didn't see any other links for contact. My indilib userid is "jimbolaya64". Please advise, thank you.

knro, 2025-04-01:
Your user is active. Maybe clear website cookies or try incognito mode and login again?

Jimbolaya64, 2025-04-01:
> Your user is active. Maybe clear website cookies or try incognito mode and login again?

Thank you for the response. Already tried clearing the cache and same issue using incognito mode.

knro, 2025-04-01:
Not sure, everything looks good for the account itself and in the forum software. For Discord, you can use this: https://discord.gg/See5qT5w

Jimbolaya64, 2025-04-01:
I logged in on my iphone using safari and can access the forums. Seems like the problem may be associated with Firefox on Debian 12. Thank you for the Discord link.

vancurad, 2025-04-02:
I had this issue yesterday, shortly after creating an account. It works today though, same browser (Safari) - if your account was new as well, it might be some transient thing.