indilib/indi #2214: Atik EFW2 no longer reporting finish of filter change

RononDex, 2025-04-03:
**Describe the bug**
When using the Atik EFW2, indi no longer realises that my Atik EFW2 finished moving the filter into place. I can physically verify that it moves the filter into place, but the INDI control panel keep showing a yellow light on the selected filter slot and the sequencer is waiting for the filter change to complete.

**To Reproduce**
Exact steps to reproduce the behavior.
1. Connect to Atik EWF2
2. Change filter through indi

The logs below clearly show that indi is seeing that the filterwheel stopped moving, but the light still stayed yellow.

**Expected behavior**
Completed filter change operation is correctly registered by EKOS / INDI

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS:
 - Version [e.g. 1.7.6, GIT..etc]

**Log Files**
```
[2025-04-03T13:13:20.944 CEST INFO ][           org.kde.kstars.indi] - Atik EFW2 :  "[INFO] Setting current filter to slot 2 "
[2025-04-03T13:13:21.097 CEST DEBG ][           org.kde.kstars.indi] - Atik EFW2 : "[DEBUG] iPosition: 0 isMoving: 1 "
[2025-04-03T13:13:22.099 CEST DEBG ][           org.kde.kstars.indi] - Atik EFW2 : "[DEBUG] iPosition: 0 isMoving: 1 "
[2025-04-03T13:13:23.102 CEST DEBG ][           org.kde.kstars.indi] - Atik EFW2 : "[DEBUG] iPosition: 0 isMoving: 1 "
[2025-04-03T13:13:24.105 CEST DEBG ][           org.kde.kstars.indi] - Atik EFW2 : "[DEBUG] iPosition: 0 isMoving: 0 "
[2025-04-03T13:13:25.108 CEST DEBG ][           org.kde.kstars.indi] - Atik EFW2 : "[DEBUG] iPosition: 0 isMoving: 0 "
[2025-04-03T13:13:26.111 CEST DEBG ][           org.kde.kstars.indi] - Atik EFW2 : "[DEBUG] iPosition: 0 isMoving: 0 "
[2025-04-03T13:13:27.114 CEST DEBG ][           org.kde.kstars.indi] - Atik EFW2 : "[DEBUG] iPosition: 0 isMoving: 0 "
```

RononDex, 2025-04-03:
I was just able to confirm that this is a driver regression, when using indi 2.1.2 and indi-3rdparty at commit ac2dbea86e79a9a2cb821bec14cbebd47c353f94 it works properly

TallFurryMan, 2025-04-03:
Thanks for this report. In your log excerpt the filter has 0 as the SDK index of the filter, even when moving. I suppose when you select another filter on your wheel this index moves to 1, 2...up to your number of filters minus one? Or it does stay at zero whatever the choice?

TallFurryMan, 2025-04-03:
Ah sorry, the copy icon was masking the first line, where you logged that the filter is the second one, thus index 1 for the SDK. I'll ping Atik on this.

RononDex, 2025-04-03:
Thanks! Thankfully it was easy for me to downgrade to a working version :)

knro, 2025-04-03:
isn't this an INDI driver issue though?

RononDex, 2025-04-24:
Do you know if this was already fixed in the current indi-3rdparty git repository?

knro, 2025-04-25:
No changes made regarding this issue. @TallFurryMan Any ideas if this was handled or noted in the latest libatik update?

TallFurryMan, 2025-04-25:
I'm still waiting for Atik here. I sent a first message on Apr 3, ping'd again one week later, and sent a mail to generic support one week ago.

RononDex, 2025-05-06:
`Which Atik camera model was used to control the EFW2 device?`

--> I do not have an Atik camera, it the filter wheel is connected via USB to my raspberry pi (over a powered usb hub).

`Which specific EFW2 device model was involved?`

--> How can I check the device model?

Regarding the other questions, I tested it with indilib 2.1.2 and the old SDK and then it worked.
I was not able to get the old indi-3rdparty with the old Atik SDK to compile with indilib 2.1.3 due to mismatch in dependencies. The drivers would instantly crash when trying to start them

TallFurryMan, 2025-05-13:
@RononDex Are you able to use your EFW on a Linux with x86_64 architecture? I can rebuild 2.1.3 with the SDK change reverted and provide it to you.

RononDex, 2025-05-13:
yes I could connect it to my desktop pc (also running arch linux, but x86_64)

TallFurryMan, 2025-05-13:
I built indi 2.1.3 with the previous Atik SDK for you to test:

indi-core: https://drive.google.com/file/d/1g48F4iWFDAf7M05Oy8fF55lJEZD7RmVb/view?usp=sharing
indi-3rdparty libraries: https://drive.google.com/file/d/1UttMAezSGuWnA1ZSE2bKP9v5dc8Wu7_B/view?usp=sharing
indi-3rdparty drivers: https://drive.google.com/file/d/1JN2B_aOtChpbG6UUAbbDzoyk1Sqwchx0/view?usp=sharing

The environment in which those files were built is Ubuntu 24.04. If that doesn't work, I can use another environment.

Of course these files are only valid for the investigation of this ticket, nothing else.

RononDex, 2025-05-14:
I will probably be able to test it in a week or so. My telescope is currently setup outside to use the good weather for imaging :)

RononDex, 2025-05-14:
Also my desktop is running on Arch Linux, I am not sure if it can install a .deb file, I might find a way. Is there no way I could compile it myself (have a separate branch I could check out?)

TallFurryMan, 2025-05-15:
Sure! https://github.com/TallFurryMan/indi-3rdparty/tree/issue/2214

knro, 2025-05-29:
Is this issue resolved? Next INDI release due in 2 days.

RononDex, 2025-05-31:
I will see if I can test it with @TallFurryMan branch tomorow

RononDex, 2025-06-01:
@TallFurryMan I just tested it using indilib 2.1.3 and your linked branch of indi-3rdparty on a x86_64 machine and there it appears to be working correctly

MarcelNoordman, 2025-06-01:
I have this problem on my machine and submitted a support ticket with Jasem.
Attached my log file for reference. I am running on a raspberry pi 4

[log_08-37-43 (1).txt](https://github.com/user-attachments/files/20542259/log_08-37-43.1.txt)

TallFurryMan, 2025-06-02:
Pinged Atik back with this. The regression is then in one of the 2024 SDKs.

TallFurryMan, 2025-06-02:
Hm, not a very good communication from them, agreed. I made it clear we were looking into the possibility of a driver regression, but here it's definitely in the SDK. And it's not really a breaking change in the communication between the indi client and the indi driver when we observe the log.

However, if we take too long to come back to Atik, we risk our ticket there to be closed with a dummy reason.

Any chance one of you could test with the ASCOM driver too?

TallFurryMan, 2025-06-04:
I got news from Atik, they ack the issue is in the SDK and will push a new version soon. No ETA yet.

RononDex, 2025-06-04:
That is good news!

MarcelNoordman, 2025-06-25:
Any progress on this issue? Do we already have an ETA?

TallFurryMan, 2025-06-25:
There is a new SDK available at Atik, but the changelog does not mention any fix related to EFW2. I'll ping Atik again.

TallFurryMan, 2025-06-29:
I got confirmation from Atik that a SDK was in QA phase to be released in the coming days, with the fix for the EFW2.

TallFurryMan, 2025-07-01:
The new SDK was posted by Atik, if anyone wants to test before I integrate.

TallFurryMan, 2025-07-06:
Don't worry I should be able to integrate the new SDK beginning of this week.

TallFurryMan, 2025-07-10:
Sorry I wanted to do too much this week. I wanted to test the new SDK on my own Atik cams, but wasn't able to. And when I started integrating the SDK in indi-3rdparty, I realised this version had no macos libraries. So I'll probably do an intermediate release for Linux, but I need to ping Atik on this issue.

TallFurryMan, 2025-07-15:
From Atik:

> Regarding the structural changes:
> 
> The FlyCapture library has been deprecated and will no longer receive any further updates or development.
> 
> We are migrating the MacOS SDK from Intel-based systems to support Apple Silicon processors (M1, M2, etc.).
> Beginning with the next SDK release, a new MacOS library will be available exclusively for Apple Silicon.
> The existing MacOS Intel library will remain available as a legacy version, but it will no longer be actively
> maintained or updated.

MarcelNoordman, 2025-07-26:
Hi team, any updates on expected date for resolution? Will this be solved with the next Ekos/Indi release? Is the problem on the server side or on the client side (or both?). I run server in Pi and Client on Mac as a remote set up.. currently I can only use one filter per night (set at start up) and need to rename and update fits header of images afterwards. Thanks for the support!

atlendor, 2025-07-27:
Hi, many thanks for all the work on this. However I just tried the update_indi_core and update_indi_drivers commands (which I *thought* pulled the updates from GitHub? But I'm no coder so maybe I have this wrong?) - anyway the filterwheel is still stuck. Not sure where I'm going wrong? (this is running on a Pi/Stellarmate by the way).

[image]

TallFurryMan, 2025-07-27:
Could you check which package is installed for the indi driver? This should tell us when it was built.

atlendor, 2025-07-27:
Hmmm not sure how to do that? is there a command? When I run update_indi_core I get things like:

---
-- The following components are going to be built:
-- ## INDI Server
-- ## INDI Drivers, Tools, and Examples
-- ## INDI Client
-- Configuring done
-- Generating done
-- Build files have been written to: /home/stellarmate/Projects/build/indi-core
[  8%] Built target sockets
[  8%] Built target dsp
[  8%] Built target indicore
[  8%] Built target eventloop
[  8%] Built target fpack
[  8%] Built target hid
[  8%] Built target indidevice
[  8%] Built target indiserver
[  8%] Built target indi_eval
[  8%] Built target indiabstractclient
[  8%] Built target indi_setprop
[  8%] Built target indi_getprop
[  8%] Built target indi_hid_test
<cut>
[  8%] Built target indi_xagyl_wheel
[  8%] Built target indi_wanderer_rotator_lite_v2
[  8%] Built target indi_ioptron_wheel
[  8%] Built target indi_simulator_wheel
[  8%] Built target indi_manual_wheel
[  8%] Built target indi_quantum_wheel
[  8%] Built target indi_qhycfw3_wheel
[  8%] Built target indi_optec_wheel
[  8%] Built target indi_qhycfw1_wheel
[  8%] Built target indi_trutech_wheel
[  8%] Built target indi_qhycfw2_wheel
[  8%] Built target indi_pegasusindigo_wheel
<cut>
---

... But I dont see an entry for atik?

atlendor, 2025-07-27:
And the log for command update_indi_drivers is attached as a file rather than clogging up this page(!) - in case this is useful.

[update_indi_drivers.txt](https://github.com/user-attachments/files/21454487/update_indi_drivers.txt)

TallFurryMan, 2025-07-27:
Given the error messages, I suggest you open a ticket to Stellarmate. It seems your indi-core diverged from the main branch, and you are missing package libftdi1-dev.

Cc @knro.

knro, 2025-07-29:
Yes please open a ticket so we can follow up with that.

atlendor, 2025-07-29:
Will-do, cheers (ticket ID 1731). But do finish your holiday/break first! (cloudy here this week anyway).

RononDex, 2025-08-04:
awesome, so I can finally update my imaging setup again without having to worry about this bug :)

RononDex, 2025-08-05:
@MarcelNoordman  As mentioned in the above comments, it is not yet in a released version, but rather fixed only if you build indilib and indi-3rdparty from source. So apt update would not be able to install the fixed version at the moment (also apt based distributions usually take quite a while to release new versions)

closed 2025-10-08 by RononDex