indilib/indi #2216: indi_manual_wheel crash since v2.1.3

zsarosi, 2025-04-04:
A couple of second after starting Indi server from EKOS / K-Stars a message pops up 'INDI driver "indi_manual_wheel" crashed!' , and it keeps popping up after closing it. Beside the message, I cannot choose now a filter manually, the list of filters is empty.
The problem appeared after the update to version 2.1.3, I haven't had issues with the older versions. 

**To Reproduce**
- configure manual filter in the EKOS profile
- start Indi from EKOS
- Error message 'INDI driver "indi_manual_wheel" crashed!' pops up after a couple of seconds
- List of filters cannot be opened / empty

**Expected behavior**
- Indi starts with manual filter enabled in the profile
- List of filters is visible, editable and not empty

**Desktop (please complete the following information):**
 - OS: Ubuntu 24.04.1
 - Version indi-bin 2.1.3+202503311639-ubuntu24.04.1
 - Kstars Version kstars-bleeding 6:3.7.6+202504031004~ubuntu24.04.1

knro, 2025-04-05:
I [triggered a build](https://code.launchpad.net/~mutlaqja/+recipe/libindi-stable) in the PPA. Please update in a couple of hours and report back.

zsarosi, 2025-04-05:
Up till now update in the repository, btw. I use this repo: https://ppa.launchpadcontent.net/mutlaqja/ppa/ubuntu/

knro, 2025-04-05:
All Launchpad builds are failing. There must be some outage on Launchpad end. Will try again tomorrow.

zsarosi, 2025-04-06:
Thanks the update to version 2.1.3+202504061737~ubuntu24.04.1 solved the issue.

closed 2025-04-06 by zsarosi