indilib/indi #2217: Main P1 Camera Stops Capture When P1 Guide Camera Loops

promero11, 2025-04-08:
**Describe the bug**
Main Capture /Download operation halts when guide camera is looping when using Ekos and Player One cameras ..
**To Reproduce**
Exact steps to reproduce the behavior.
1. Start EKOS & INDI Server
2. Press "start" arrow in EKOS camera tab. Press Loop button in EKOS Guide tab
3. Main P1 POSIEDON-M camera halts capture/download process when zP1 Ceres autoguider loops

**Expected behavior**
No effect on capture process.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Kubuntu 24.04
 -Latest

[indi_app_log_25-04-07T19-01-56.txt](https://github.com/user-attachments/files/19641608/indi_app_log_25-04-07T19-01-56.txt)

[indi_debug_log_25-04-07T19-01-56.txt](https://github.com/user-attachments/files/19641612/indi_debug_log_25-04-07T19-01-56.txt)

[kstars_logs_2025-04-07T19-07-02.zip](https://github.com/user-attachments/files/19641621/kstars_logs_2025-04-07T19-07-02.zip)

[indi_logs_25-04-07T19-01-56.zip](https://github.com/user-attachments/files/19641622/indi_logs_25-04-07T19-01-56.zip)

**Log Files**
Make sure to enable logging and include log files https://indilib.org/support/logs-submission.html

knro, 2025-04-08:
I will trigger a rebuild of libplayerone and indi-playerone in the PPA.

promero11, 2025-04-09:
Hi Jasem, Thanks! What do I have to do, if anything? 
Clear Skies,

Paul

knro, 2025-04-10:
Looks like the build for 24.04 failed: https://code.launchpad.net/~mutlaqja/+recipe/indi-playerone-stable

I triggered it again. If it is successful, you can update again and check.

promero11, 2025-04-14:
Hi all,

I did an "update_indi_core" and "update_indi_drivers" in the konsole, then started Ekos.
The same thing is happening: capture stops when guider is set to looping. I looked at the fits headers in the Ekos viewer for both the main and the guider camera panels

[log_19-10-32.txt](https://github.com/user-attachments/files/19728852/log_19-10-32.txt)

[log_19-00-27.txt](https://github.com/user-attachments/files/19728855/log_19-00-27.txt)
[log_19-10-32.txt](https://github.com/user-attachments/files/19728854/log_19-10-32.txt)

, and the cameras are the same camera (both main and guider) as evidenced by the pixel size of the output. In my imaging train, I state Poseidon _M Pro as main camera and Xena as guide camera. In my profile, I only have one: indi_playerone_ccd, and when I start Ekos, the Indi Control Panel lists all three of my Player One cameras as successfully connected (2x Posiedon and 1x Xena) The serial numbers shown in the Indi Control Panel are: Poseidon Main #1: CAMDA223B85002109000, Poseidon Main (2nd piggybacked scope) #2: CAMD5322D85002109000, and Xena Guide Cam: CAMG50227DA041109000, so it does appear the driver is connecting to all three cameras. My next step is to remove one of the telescopes and just do a single main camer and guide scope. I have attached the logs.

Clear Skies, Paul[](url)

knro, 2025-04-14:
@hiro3110i I'm not sure if you can recommend anything?

promero11, 2025-04-14:
Hi guys,

A quick search of the internet reveals this problem affects every OS. I have already replaced my USB cabkes and the back camera plates as part if a warranty. When I found this post on Astrobin, I think it surmises the problem with the drivers:

 “I use a Player One Xena-M for guiding and a Player One Ares-M for imaging with the SCT.  I had the exact same issue: there were two ASCOM drivers for the cameras and each showed both cameras. I deleted ASCOM 2 and removed the Ares-M from ASCOM 1. ASCOM 1 now only runs the Xena-M and the Ares-M is run only with the PO proprietary driver.  It all appears to be working properly now.”

I know the INDI drivers were working in the neginning, but I am unfamiliar with Linux and dont know how to completely remove the Player One Driver and re-install. Could you help?

CleR Skies,

Paul

knro, 2025-04-15:
If using PlayerOne Camera 1 and Camera 2 drivers, same issue?

hiro3110i, 2025-04-18:
@promero11 Thank you for reporting your issue.

Could you please check optical train settings of Kstars and optical train selection of EKOS?
A similar phenomenon I experienced was when the EKOS optical train selection was the same in the CCD tab and the Guide tab.
I didn't notice this for a while, and it took me a while to resolve the problem.
We must select the appropriate camera in each optical train settings and EKOS tab.
Capture will stop if there is a camera conflict.
If you use two cameras of the same model, be careful as they will have the same camera names and it is easy to make a mistake.

Another consideration is USB bus power.
Poseidon consumes a lot of power, so the USB power supply capacity must be high enough.
This can be tested by using an externally powered USB hub.

promero11, 2025-04-21:
[image]

[image]

[image]

Hi hiro3110i,

Thanks for your troubleshooting help. I finally got to look at it this weekend. The problem persists.
Things I've tried:
1) Checked optical train selections (found out I had the wrong filterwheel in the second optical train)
2) Physically disconnected one of the Poseidon M cameras.
3) Physically disconnected the other Poseidon M camera and in the software changed  the Poseidon M camera in  the primary optical train.
4) Restarted PC on each step above.
5) Checked my Power supply, and it is a Pyramid 14 amp newer unit. The telescope rig pulls less than 1 amp at idle and even with one of the Poseidon M cameras cooling, it is less than 2 amps. Voltage is at 13.87 volts.
6) Connected all Poseidon M cameras to a powered StarTech USB 3 hub, instead of the unpowered hub they were on.

As far as USB power, the only potential problem could be an off brand power splitter that powers everthing on the telescope, with the StarTech powered USB 3 hub just being one of many. I will replace this unit. Do you have a recommendation? I am using "Anderson Powerpole" type connectors and need a model that has that type.

I look forward to working on this further,

Clear Skies,

Paul

promero11, 2025-04-21:
hiro3110i,

I reread your post, especially: “ Could you please check optical train settings of Kstars and optical train selection of EKOS?
A similar phenomenon I experienced was when the EKOS optical train selection was the same in the CCD tab and the Guide tab.”

The first thing that came to mind us that the optical train in the guide tab should be it’s own unique one? I will attempt to create a new optical train when I can get to it and report back.

Clear Skies,

Paul

promero11, 2025-04-22:
hiro3110i,

Your explanation works great on my setup. Thank You for y our time!

Clear Skies,
Paul

closed 2025-04-22 by promero11

hiro3110i, 2025-04-23:
@promero11 I'm so sorry for my late response.
I'm glad the problem was solved.

Clear Skies, too