indilib/indi #2225: iOptron CEM25 Compatibility with the iOptronV3 INDI Driver

Trigger-broom-289, 2025-04-18:
This is a follow up from the following [closed issue](https://github.com/indilib/indi/issues/1542)
I do have a CEM25p and it does connect with the V3 driver. However I'm getting the following messages in the info box.

```
2025-04-18T15:07:00: [ERROR] bool IOPv3::Driver::getStatus(IOPv3::IOPInfo*): Expected 23 bytes but received 19.
2025-04-18T15:07:00: [ERROR] Read Command Error: Timeout error

```

The driver https://github.com/indilib/indi/blob/master/drivers/telescope/ioptronv3driver.cpp does not show the CEM25(p) in the list.

There is also the following which I do not understand
```
if (strlen(res) != 23)
    {
        DEBUGFDEVICE(m_DeviceName, INDI::Logger::DBG_ERROR, "%s: Expected 23 bytes but received %d.", __PRETTY_FUNCTION__,
                     strlen(res));
        return false;
    }
```

I would like to be able to use the PEC function.
I'm Gonzothegreat on the indi forum and radada on the Stellarmate Discord.

thanks

Trigger-broom-289, 2025-04-18:
My mount controller, the DEC and RA boards are fully up to date according to the iOpton website support page.

knro, 2025-04-19:
Because the expected response from the getStatus command is 23 bytes. What's the response you get from yours? There has been reports that sometimes two commands receive a response *out of order* which might cause this. A more complete log can confirm this.

Trigger-broom-289, 2025-04-19:
I receive 19, I will provide a log file at some point today.

Trigger-broom-289, 2025-04-21:
Here's the log file from today, interesting that the mount is being seen as a HEM47.
The mount isn't slewing correctly, you hit any of the directions and it moves however it doesn't stop when you release the command.

[log_12-18-15.txt](https://github.com/user-attachments/files/19833003/log_12-18-15.txt)

Trigger-broom-289, 2025-04-21:
From the log file, I can see the following:
```
[2025-04-21T12:21:09.685 CEST DEBG ][           org.kde.kstars.indi] - iOptron v3 : "[DEBUG] Getting firmware data... "
[2025-04-21T12:21:09.685 CEST DEBG ][           org.kde.kstars.indi] - iOptron v3 : "[DEBUG] CMD <:MountInfo#> "
[2025-04-21T12:21:09.693 CEST DEBG ][           org.kde.kstars.indi] - iOptron v3 : "[DEBUG] RES <0025> "
```

And in the driver https://github.com/indilib/indi/blob/master/drivers/telescope/ioptronv3driver.cpp
<0025> is HEM27

knro, 2025-04-21:
Maybe it's not the correct driver for it? It appears to not respond at all to **:GEP#** command. If you use cutecom for example, can you get the mount to respond to this? It's responding to :GLS# command apparently.

Trigger-broom-289, 2025-04-21:
In https://indilib.org/devices/telescopes/ioptron/cem120.html we can see that this driver support the CEM25(p)
CEM25(/P) with firmware 20171001 and later

Trigger-broom-289, 2025-04-21:
As for the :GEP# command, could it be that the RS232 Version is different on the CEM25?
V2.0 vs V3.0

Trigger-broom-289, 2025-04-21:
I need to check https://www.ioptron.com/v/ASCOM/RS-232_Command_Language2014_V2.5.pdf

Trigger-broom-289, 2025-04-21:
GEP is not available in the v2.5 pdf
however it is in V3.1 https://www.ioptron.com/v/ASCOM/RS-232_Command_Language2014_V2.5.pdf

Trigger-broom-289, 2025-04-21:
In version v2.5 the command appears to be :GEC# instead of :GEP#

In V3.10 we have the following notice:
```
**Added get current position command “:GEP#”, this command now supersedes “:GEC#”. “:GEC#”
has been removed.**
```

Trigger-broom-289, 2025-04-21:
I might modify https://github.com/indilib/indi/blob/master/drivers/telescope/ioptronv3driver.cpp#L756 and see how it goes.

Trigger-broom-289, 2025-04-21:
regarding:
```
iOptron v3 : "[DEBUG] CMD <:MountInfo#> "
iOptron v3 : "[DEBUG] RES <0025>
```

In **V2.5**
Command: “:MountInfo#”
Response: “0010”, “0011”, “0025”, “0026”, “0030”, “0040”, “0041”, “0045”, “0046”, “0060”,
“0061”, “5010”, “5035”
This command gets the mount model. “0010” means Cube II or Cube Pro EQ mode, “0011” means
SmartEQ Pro+, **“0025” means CEM25(/P)**, “0026” means CEM25-EC, “0030” means iEQ30 Pro,
“0040” means CEM40, “0041” means CEM40-EC, “0045” means iEQ45 Pro EQ mode, “0046”
means iEQ45 Pro AA mode, “0060” means CEM60, “0061” means CEM60-EC, “5010” means
Cube II or Cube Pro AA mode, “5035” means AZ Mount Pro.

knro, 2025-04-21:
Ok thank you for the follow up! Let me check why there is a discrepancy in the model names.

knro, 2025-04-21:
Alright, so it appears we have two model lists, one for the [older drivers](https://github.com/indilib/indi/blob/529060083e8e3dfac90fed3538aa6b4d4bc97500/drivers/telescope/ieqdriverbase.h#L239) and another list for the [latest v3.1 mounts](https://github.com/indilib/indi/blob/529060083e8e3dfac90fed3538aa6b4d4bc97500/drivers/telescope/ioptronv3driver.cpp#L44). Need to find the source for the HEM27 information.

Trigger-broom-289, 2025-04-21:
I'm currently using the indi_ieq_telescope driver for my CEM25p, but it's missing quite a bit from the V3 driver, like PEC.

Trigger-broom-289, 2025-04-21:
> Alright, so it appears we have two model lists, one for the [older drivers](https://github.com/indilib/indi/blob/529060083e8e3dfac90fed3538aa6b4d4bc97500/drivers/telescope/ieqdriverbase.h#L239) and [...]

The source for HEM27 is as I've described, MountInfo returns 0025 as I'm trying to use the V3 drivers it gives you HEM27.

Trigger-broom-289, 2025-04-21:
I've updated the link in https://github.com/indilib/indi/issues/2225#issuecomment-2818240086

Trigger-broom-289, 2025-04-21:
And there's the link to the RS-232 V3
https://www.ioptron.com/v/ASCOM/RS-232_Command_Language2014V3.pdf

Trigger-broom-289, 2025-04-21:
The error "Expected 23 bytes but received 19." comes from https://github.com/indilib/indi/blob/master/drivers/telescope/ioptronv3driver.cpp#L293

RS-232 V2.5:
Command: “:GLS#”
Response: “sSSSSSSSSSSSSnnnnnn#”
This command gets longitude, latitude and all kinds of status. The response includes a sign and 18 digits.

RS-232 V3:
Command: “:GLS#”
Response: “sTTTTTTTTTTTTTTTTnnnnnn#”
This command gets longitude, latitude and all kinds of status. The response includes a sign and 22 digits.

Trigger-broom-289, 2025-04-21:
I'm currently making some modifications to the V3 driver to include the CEM25p, will let you know if I'm successful or not...

Trigger-broom-289, 2025-04-21:
I think this is where the modification could end... 
https://github.com/indilib/indi/blob/master/drivers/telescope/ioptronv3driver.cpp#L453C1-L480C14

The RS-232 V2.5 does not have ":GPE#" nor ":GPR#"

Command: “:GPE#”
Response: “0”, “1”
This command gets the integrity of the periodic error data. “1” means complete, “0” means incomplete.
Note: This command only available in non-encoder version of equatorial mount.

Command: “:GPR#”
Response: “0”, “1”
This command gets the status of PE recording. “0” stands for PE recording stopped, “1” stands for PE is being recording.
Note: This command only available in non-encoder version of equatorial mount.

Trigger-broom-289, 2025-04-21:
mmm interesting...

**V3.10:**
iOptron® Mount RS-232 Command Language
Version 3.10
January 4th, 2021

**V3:**
iOptron® Mount RS-232 Command Language 2014
Version 3.0
November 26th, 2017

**V2.5:**
iOptron® Mount RS-232 Command Language 2014
Version 2.5
January 15th, 2019

V3 is older than V2.5

Trigger-broom-289, 2025-04-21:
I think we can close this issue, the iOptron V3 driver will not work on the CEM25(p) due to:
indi iOptron driver V3 only support mounts with RS-232 V3.x
the CEM25p mount runs RS-232 V2.x

What I recommend is that the following page https://indilib.org/devices/telescopes/ioptron/cem120.html gets updated to avoid further confusion.
We need to remove the following lines:
```
CEM25-EC with firmware 20171001 and later
CEM25(/P) with firmware 20171001 and later
```

The correct list is at https://www.ioptron.com/Articles.asp?ID=311

closed 2025-06-12 by Trigger-broom-289