indilib/indi #2239: wave150i and indi_eqmod driver

coutanto, 2025-05-20:
**Describe the bug**
The position of the mount reported by INDI has a +6h shift on RA with respect to the position reported by skywatcher synscan app. When pointing to a target and shooting the same shift is observed. The observed target is 6h RA ahead, while the fits header contains the expected RA,DEC position.
This is observed for both kstars/ekos, and direct PyIndi request.

**To Reproduce**
Exact steps to reproduce the behavior.
1. Run driver
2. set a target to be pointed (RA, DEC) either by INDI (python or ekos), or by skywatcher SynScan proprietary app.
3. Once the mount is orientated and folow the target, read the mount position
4. The position reported by INDI as an AD with a 6h shift with respect to SynCan reported position.
5. The same behavior was observed during a regular observation. The fits image coordinates were the 
    one requested, but the image was pointing to a coordinates shifted by 6h

**Expected behavior**
The position should be identical

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: ubuntu 22.04.5 LTS on i486; ubuntu 22 on raspbery
 - Version [2.1.0 git]

**Log Files**
Make sure to enable logging and include log files https://indilib.org/support/logs-submission.html

knro, 2025-05-20:
I can imagine the +6 Hour shift only when the mount is parked, but not when it is tracking an object after slew. You slewed to a star and it reported this difference? If yes, please send logs as this is highly unusual.

coutanto, 2025-05-23:
Almost sure that this bug is not due to indi. I suspect kstars/ekos to be responsible for it.
Test:
- indiserver running indi version github 2.1.2 github  on a remote server
Then:
- I try an old version of kstars/ekos (kstars 3.5.7 Stable) and this problem doesn't exist
- I try  recent versions (kstars 3.7.7 Beta or kstars 3.7.5 Stable) and the problem does exist
I switch to kstars/ekos github.

closed 2025-05-23 by coutanto