indilib/indi #2240: INDI and Siril cannot be installed via repositories at the same time on Ubuntu 24.04

setnes, 2025-05-20:
When installing from the repositories, INDI requires libxisf (provided by INDI PPA), and Siril requires libxisf0 (provided by Ubuntu).  These two packages conflict and cannot be installed at the same time.

TallFurryMan, 2025-05-21:
Is it that the two packages conflict, or that the dependency spec of indi only targets libxisf? Which version of Ubuntu do you see the problem on?

setnes, 2025-05-21:
Ubuntu 24.04.2 LTS

When trying to install Siril...

```
The following packages have unmet dependencies:
 libxisf : Conflicts: libxisf0 but 0.2.8-1 is to be installed
E: Error, pkgProblemResolver::Resolve generated breaks, this may be caused by held packages.
```

setnes, 2025-06-11:
This might be fixed soon.  I assume we need to wait for the new builds to roll out in the PPA.

https://bugs.kde.org/show_bug.cgi?id=503146#c8

knro, 2025-06-11:
Ok so it will not be fixed soon. The Ubuntu's version of libxisf is too old. What you can do is install [KStars Flatpak](https://indilib.org/forum/ekos/15496-kstars-flatpak-testing.html) and then install Siril from PPA.

setnes, 2025-06-11:
I might try the Siril flatpack instead.  I don't use KStars, just INDI.  I also don't use PixInsight, so needing to mess around like this just to avoid the xisf dependencies is a bit frustrating.

setnes, 2025-08-20:
For what it's worth, I've been using the Siril flatpak.  This allows me to use the INDI PPA without installing Kstars/Ekos.  I'm sure Ekos is great, I just really like CCDCiel, so I use that instead.