indilib/indi #2247: celestrongps.cpp with Nexstar 6SE mount: improperly supported/unsupported features

dmallett03, 2025-06-18:
The indi_celestron_gps driver reports that the "mount firmware does not support track mode" and "mount firmware does not support getting peer side".
These messages are shown in kstars' Indi Control Panel's message window when starting the service (example: Run Celestron Nexstar HC Service in kstars).
I am not sure about the peer side, but the mount does support tracking, and I would expect it to show up as supported.
This affects indi_celestron_gps version 3.6 (and many previous ones I've tried). I am using ubuntu 16.04.
Additionally, the Nexstar 6SE mount can correctly track an object by default, however using the Indi Control Panel's Motion Control buttons (as to center the object) correctly moves the mount, but the mount stops tracking and drifts.
One way to restore tracking is to disable and re-enable tracking in the INDI Control Panel's Main Control page.
The code shows exceptions for many models, but my guess is that the driver has been written for an older "Nexstar" model (as per before the release of the Nexstar xSE).
I can test corrected versions of the code if need be.

knro, 2025-06-19:
Ubuntu 16.04? Are you sure about that?

dmallett03, 2025-06-27:
@knro  : Thank you very much for looking at this issue. In the meantime, in order to have a semi-working setup, I use a workaround where I force a Tracking Enable call at then end of the Sync routine in the driver (celestrongps.cpp). My working sequence is: align mount (tracking is enabled by default), move to the object I want to observe (through KStars), center the object remotely through the INDI Control Panel's Move buttons, and once the object is centered, I select (KStar's Celestron Telescope's right-click) "Sync" option to have the object and KStar's crosshairs match. There, as mentioned above, I added a Tracking Enable call at the end of the Sync function, so tracking is restored after the Sync. Since this is a cheap and dirty work-around, it eventually stops working (after about after 5 object change sequences described above), so I'm looking forward to a permanent, more robust soution.

dmallett03, 2025-09-05:
Why has this isdue been closed? It was never adressed.

knro, 2025-09-05:
It was closed due to inactivity. No developers picked up the issue, and I don't have time for it. If you can inspect the code and find the issue, or at least point to where it can be then maybe we can look further into it.

dmallett03, 2025-09-16:
Thanks for your reply. 
In order to better understand the code, could you please provide the sources for the information used to select the mount's features below ?

The changes below allow my mount to resume tracking after a manual displacement or a goto operation, but I'd need more information to better implement a solution.

I am using the celestrongps.cpp file version described as:
setVersion(3, 6); // update libindi/drivers.xml as well

In the following code snippet, my mount does not return a version higher than 3.2. 
Is version 3.2 valid for a Nexstar SE, or is it for another (previous, maybe?) version of the Nexstar ?
As it is now, i get a message saying that tracking and sync are not supported.

bool CelestronGPS::checkMinVersion(double minVersion, const char *feature, bool debug)
{
    if (((fwInfo.controllerVariant == ISSTARSENSE) &&
            (fwInfo.controllerVersion < MINSTSENSVER)) ||
            ((fwInfo.controllerVariant == ISNEXSTAR) &&
             (fwInfo.controllerVersion < minVersion)))
    {
        if (debug)
            LOGF_DEBUG("Firmware v%3.2f does not support %s. Minimum required version is %3.2f",
                       fwInfo.controllerVersion, feature, minVersion);
        else
            LOGF_WARN("Firmware v%3.2f does not support %s. Minimum required version is %3.2f",
                      fwInfo.controllerVersion, feature, minVersion);

        return false;
    }
    return true;
}

My (cheap and dirty) work-around is to do the followng:

In the following part, I had to force the enabling of the Sync feature. 
I kept your comment a few lines above the change I made. Maybe it's related. 
Where did you get the information saying the minimum version to enable sync is 4.1 ?

        // JM 2018-09-28: According to user reports in this thread:
        // http://www.indilib.org/forum/mounts/2208-celestron-avx-mount-and-starsense.html
        // Parking is also supported fine with StarSense
        if (checkMinVersion(2.3, "park"))
            cap |= TELESCOPE_CAN_PARK;

// Force enabling Sync
//        if (checkMinVersion(4.1, "sync"))
if (checkMinVersion(1.0, "sync"))
            cap |= TELESCOPE_CAN_SYNC;

Same here. Where did we get version 2.3+ for track mode ?  
my mount does not report version 2.3+ or better, but tacking is definitely supported.

        // Track Mode (t) is only supported for 2.3+
        CELESTRON_TRACK_MODE ctm = CTM_OFF;
// DM: Allow Track Mode
//        if (checkMinVersion(2.3, "track mode"))
if (checkMinVersion(1.0, "track mode"))

In the following, I also had to force track mode when in Alt-Az, een though my mount can track even in Alt-Az mode. 
Wondering why track mode would be disabled in Alt-Az...

//        if (fwInfo.celestronTrackMode != CTM_ALTAZ)
if (1)
            cap |= TELESCOPE_HAS_TRACK_MODE;
        else
...

Below, I had to force Alt-Az tracking. 

        //GUIDE Update properties.
        // check if the mount type and version supports guiding
        // Only show the guide information for mounts that
        // support guiding.  That's GEMs and fork mounts in equatorial modes.
        // well, anything in an equatorial mode
        if (fwInfo.celestronTrackMode == CELESTRON_TRACK_MODE::CTM_EQN ||
                fwInfo.celestronTrackMode == CELESTRON_TRACK_MODE::CTM_EQS ||
// DM: Allow Alt-Az tracking:
fwInfo.celestronTrackMode == CELESTRON_TRACK_MODE::CTM_ALTAZ ||
                fwInfo.celestronTrackMode == CELESTRON_TRACK_MODE::CTM_RADEC)

Below I added a call to unsync at the beginnning of the Goto call, as recommended in the Celestron manual:

bool CelestronGPS::Goto(double ra, double dec)
{
     targetRA = ra;
     targetDEC = dec;
// Unsync:
driver.unsync();
...

And force tracking back on at the end of the CelestronGPS::Sync call

...
SetTrackEnable(true);
TrackState = SCOPE-TRACKING;
     return true;
}

knro, 2025-09-16:
What version is reported by your mount?

knro, 2025-09-17:
Thank you for the detailed response. This is odd, because in the code, we have this:

```
bool CelestronGPS::checkMinVersion(double minVersion, const char *feature, bool debug)
{
    if (((fwInfo.controllerVariant == ISSTARSENSE) &&
            (fwInfo.controllerVersion < MINSTSENSVER)) ||
            ((fwInfo.controllerVariant == ISNEXSTAR) &&
             (fwInfo.controllerVersion < minVersion)))
    {
        if (debug)
            LOGF_DEBUG("Firmware v%3.2f does not support %s. Minimum required version is %3.2f",
                       fwInfo.controllerVersion, feature, minVersion);
        else
            LOGF_WARN("Firmware v%3.2f does not support %s. Minimum required version is %3.2f",
                      fwInfo.controllerVersion, feature, minVersion);

        return false;
    }
    return true;
}
```

So the log should **indicate** what firmware version it detected versus the firmware version required, but this is not shown in your logs. This could mean your installation of INDI is probably old. You might have a rouge INDI installation under /usr/local as well.

dmallett03, 2025-09-18:
Hello and thanks for your reply.
The INDI installation and driver have been recompiled.
I can change the version of the driver number from 3.6 to 3.7 and the version displayed in KStars is 3.7, so no rogue driver.
Please note that the NexStar telescope series has been around for a while, and the NexStar 6/8 SE (for Second Edition, I guess) might be more recent than the one used when writing the Driver. Possibly the firmware version structure has changed between NexStar versions (?). 

The FirmwareInfo from celestrondriver.h looks ike below.
Note the commented line.

typedef struct
{
    std::string Model;
    std::string Version;
    //std::string GPSFirmware;
    std::string RAFirmware;
    std::string DEFirmware;
    double controllerVersion;
    char controllerVariant;
    bool isGem;
    bool canPec;
    bool hasHomeIndex;
    bool hasFocuser;
    CELESTRON_TRACK_MODE celestronTrackMode;
} FirmwareInfo;

I uncommented the "//std::string GPSFirmware;"  and read back: 

2025-09-18T00:03:19: [INFO] Updating location is not necessary since mount is already aligned. 
2025-09-18T00:03:19: [INFO] Updating time is not necessary since mount is already aligned. 
2025-09-18T00:03:19: [INFO] Mount UTC offset: -3.00. UTC time: 2025-09-16T23:19:39. DST: On 
2025-09-18T00:03:19: [INFO] Mount supports guiding. 
2025-09-18T00:03:19: [INFO] Mount is unparked. 
2025-09-18T00:03:19: [INFO] Mount is unparked. 
2025-09-18T00:03:19: [WARNING] Mount firmware does not support getting pier side. 
2025-09-18T00:03:17: [INFO] Mount model: 6/8 SE 
2025-09-18T00:03:17: [INFO] Controller version: 5.35 
2025-09-18T00:03:17: [INFO] Celestron NexStar HC is online. 

Which makes a bit more sense.

I'll need a bit of clear sky to verify the behavior.
I'll return with results.
Regards,
Dan

knro, 2025-09-18:
I don't see the warning about **Track Mode** in either log?

dmallett03, 2025-09-18:
[Indi_Celestron_Gps_Tracking.zip](https://github.com/user-attachments/files/22414816/Indi_Celestron_Gps_Tracking.zip)

My aologies. I left the modifications (described in an earlier message) to enable Sync and Track Mode in celestrongps.cpp.

I now have replaced the celestrongps.cpp file with the original and generated 2 new logs.

indi_celestron_gps_19:49:44_Commented.log: 
this is the log file with the original celestrondriver.h with the commented out "//std::string GPSFirmware;" line.

indi_celestron_gps_19:27:04_Uncommented.log:
This is the log file with the uncommented "std::string GPSFirmware;" line.

To see if the commented/uncommented line in celestrondriver.h file had any effect, I also generated logs with the modifications with both the commented and uncommented line in celestrondriver.h.

The warning is present in both cases.
WARNING	19.414419 sec	: Mount firmware does not support track mode.
So the commented/uncommented line does not seem to have much of an effect, but I find it suspicious that the entry causing the problem is on that line.

Why are we getting this warning, then ?

For the sake of completeness, I added the logs with the modification (described earlier) that forces Track Mode.

indi_celestron_gps_20:18:49_UncommentedPlusMods.log:
As the title says, uncommented plus modifications.

indi_celestron_gps_20:42:41_CommentedPlusMod.log:
And commented plus modifications.

knro, 2025-09-19:
Please check the update driver and let me know if it fixes the issue.

closed 2025-09-19 by knro

dmallett03, 2025-09-19:
Thank you very much. I'll do that as soon as I figure out how to use "pull".

dmallett03, 2025-09-30:
We finally had a bit of clear sky, so I was able to test the updated driver. 
I am very happy to say that I was able to correctly track objects. 
Note that there is a small drift after fine positioning with the Indi panel movement buttons, but correct tracking can be restored by right-clicking the object and selecting the Celestron Nexstar's Sync option. 
Thank you very much for your great help.

commit 6684fe9f (2025-09-19): Do not confuse Celestron tracking mode with INDI tracking mode (which is used for tracking rates switch). Fixes #2247